prusa3d/PrusaSlicer #9723: Organic supports for a second instance print over the first instance

luciengrise, 2023-02-14:
### Description of the bug

Expected result:  When multiple instances or objects are on the build plate, PrusaSlicer would create "shared" tree supports that branch out to each object.  
Actual result: PrusaSlicer creates unique supports for each object which intersect with the other object.  Preview shows a set of tree supports for each object overlapped in the same x/y location as the other object's position.  The Preview timeline shows them printing sequentially:  the first object's supports, then the first object, then the second object's supports, then the second object.  This isn't just a first layer issue, as the overlap occurs again in higher layers (up to at least layer 20 in attached example).  I didn't actually print this because having the print nozzle run into existing layers seems like a bad idea.
Workaround:  The user can avoid the overprint issue by manually repositioning the second object further away, but PrusaSlicer shouldn't overlap print commands in the same location, or should at least provide a warning.
Error messages:  none
[Screenshots.zip](https://github.com/prusa3d/PrusaSlicer/files/10736812/Screenshots.zip)
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/10736813/config.zip)
[System info.zip](https://github.com/prusa3d/PrusaSlicer/files/10736814/System.info.zip)

### Project file & How to reproduce

Open a .3mf file, add paint-on supports using Automatic painting and Organic supports (that start on the build plate outside the object's perimeter), then add a second instance and click Arrange.  Slice the project, then preview slices starting with Layer 1.
[Quilt grip final v8.zip](https://github.com/prusa3d/PrusaSlicer/files/10736780/Quilt.grip.final.v8.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Windows 10 Pro, Version 21H2

### Printer model

Prusa Mini+

neophyl, 2023-02-14:
Supports and objects have always done that in PS.  In all versions and with all support types.  You have to merge the separate objects into one for that not to happen.  Its just more noticeable now as organic supports extend further from the object.

This is reported very regularly and closed as duplicate / intended operation.

lukasmatena, 2023-02-15:
Duplicate of #316. Closing.

closed 2023-02-15 by lukasmatena