prusa3d/PrusaSlicer #9729: Access Violation Error

3Djoe69, 2023-02-15:
### Description of the bug

When ever I slice the attached 3mF file I get a Access Violation Error
This is the only file that causes this to my knowledge my other files seem to work.

### Project file & How to reproduce

I just open in slicer and hit slice and the error appears

[Petes 1st Place v2.zip](https://github.com/prusa3d/PrusaSlicer/files/10745342/Petes.1st.Place.v2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 3

### Operating system

Windows 10 (10.019044 build 19044)

### Printer model

MK3S+

lukasmatena, 2023-02-15:
I cannot reproduce this in neither 2.6.0-alpha2 nor 2.6.0-alpha3.

3Djoe69, 2023-02-15:
Not sure why. I even deleted my current 2.6 A3 files and redownloaded and tried again same error.

3Djoe69, 2023-02-16:
Iv narrowed it down to the print settings I had it on. Switch print settings profile and it sliced... Not sure why this profile decided to be weird but happy its resolved thanks.

lukasmatena, 2023-02-16:
If you can still reproduce the problem, could you please zip your configuration folder (Help- >Show Configuration Folder), make sure that just opening PS and slicing the 3mf crashes and send both to us?

sniger7700, 2023-02-16:
i got the same issue some time..

sniger7700, 2023-02-16:
i removed the organic support settings. and did go back to grid, and then it work. but still. i dont need support. on this part 

Active support under settings, set it go grid,. and then slice without support . then it worked for me.

3Djoe69, 2023-02-17:
Here ya go I confirmed it still shows error when slicing with organic supports (even though it doesnt need supports its the profile I mainly use rn)

[Config.zip](https://github.com/prusa3d/PrusaSlicer/files/10767257/Config.zip)

lukasmatena, 2023-02-17:
@3Djoe69 Thanks. I managed to reproduce the crash with your config folder in 2.6.0-alpha3. However, it seems to no longer crash in 2.6.0-alpha4, which will be released later today. It is probably a duplicate of something we already fixed, maybe of #9557. Please, retest in alpha4 when it is out.

I am unable to reproduce the crash in 2.6.0-alpha3 with fresh config directory, which is a bit surprising to me.

lukasmatena, 2023-02-17:
@3Djoe69 Ok, I see it now. Your 3MF is not a PrusaSlicer project, but it is generated by some other software. Therefore it does not contain your slicing settings, which I did not notice before. Ok, so it was not that mysterious after all.

@sniger You did not provide your project, but it is very likely that your issue will be solved in 2.6.0-alpha4 as well. Let us know if it isn't.

Closing.

closed 2023-02-17 by lukasmatena