prusa3d/PrusaSlicer #9739: Organic supports don't work with adaptive layer height

jonathanalevi, 2023-02-16:
### Description of the bug

If you turn on adaptive layer height, organic supports don't scale proportionally: 
[image]

### Project file & How to reproduce

[mother_mold_for_thickness_2.6mm-3mm.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10753162/mother_mold_for_thickness_2.6mm-3mm.3mf.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 Alpha

### Operating system

MacOS 13.1

### Printer model

Voron stealth burner.

neophyl, 2023-02-16:
Correct, that's mentioned in the release notes.  Its been reported so so many times here that the alpha 3 release specifically has a block on not allowing that combination because so many people seem unable to read those release notes.

jonathanalevi, 2023-02-16:
Correct, we do not read release notes.

closed 2023-02-16 by jonathanalevi