prusa3d/PrusaSlicer #9744: Incorrect warning message

jaslo, 2023-02-16:
### Description of the bug

Print a simple object -- got "Detected print stability issues" warning
"floating object part"
[cup cage base.zip](https://github.com/prusa3d/PrusaSlicer/files/10760967/cup.cage.base.zip)

### Project file & How to reproduce

[cup cage v4.zip](https://github.com/prusa3d/PrusaSlicer/files/10761009/cup.cage.v4.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha 3

### Operating system

Windows 10

### Printer model

Prusa MK3S

Godrak, 2023-02-17:
Thank you for the report. We know about the issue, it sometimes happens on simple models with straight walls, where the detection of overlap of two consecutive layers fails. We are working on a fix.

lukasmatena, 2023-04-02:
Fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena