prusa3d/PrusaSlicer #9748: Define the slicer layout

LexHouben, 2023-02-17:
I would like to propose an improvement for the slicer feature when using a widescreen monitor. Currently, when attempting to view the print time after calculation, I am required to scroll down after each slice. This becomes particularly cumbersome when positioning multiple objects, as the position of the vertical scrollbar is lost.

Therefore, I suggest implementing an option to configure the (right) section of the slicer. One option could be to allow the vertical scrollbar to remain in the position it was prior to beginning calculations. Alternatively, the object overview window could be made smaller if the former option is not possible.

I believe that implementing this improvement would make the slicing process more efficient and improve user experience.

Thank you for considering my suggestion.
[image]
[image]

lukasmatena, 2023-02-17:
>  when attempting to view the print time after calculation

In 2.6.0-alpha3, the time is also shown in the legend window.

LexHouben, 2023-02-17:
Where?
[image]

lukasmatena, 2023-02-17:
When you switch to preview:
[image]

LexHouben, 2023-02-23:
Hi, 
Thanks for the answer but it no workable answer. Why you can try it yourself. Place some objects on the plate en turn and more them around.
To see the time:
- goto slice preview
- and then select your idea 
- then I see the time!
 
But if I want to move an object again:
- go back to the 3d view
- move/turn the object
- goto slice preview
- and see the time.

So to many clicks for something that is already available in the 3d view but not visible until you use the vertical slide bar.
Easy solution:  **Make the object overview window smaller and save the size in the settings.** Problem solved!

closed 2026-09-29 by prusaslicer-bot[bot]