prusa3d/PrusaSlicer #9749: Bridge Fan Speed is not applied to overhangs

wuast94, 2023-02-17:
### Description of the bug

Bridge Fan Speed is not applied to overhangs

also internal overhang perimeters should be slowed down or a setting should be available for this

### Project file & How to reproduce

[3DBench.zip](https://github.com/prusa3d/PrusaSlicer/files/10766046/3DBench.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3

### Operating system

Win11

### Printer model

Ratrig 500³

kubispe1, 2023-02-17:
Thanks, we would like to focus also on this topic.

dansity, 2023-02-19:
Just today I have opened a topic on the community forum with the same issue. Glad to see I'm not the only one with the problem. Additional screenshots in the topic
https://forum.prusa3d.com/forum/prusaslicer/fan-speed-not-applied-to-overhang/

Gunner087, 2023-02-20:
Might be nice to have a dynamic fan speed option too that works like the dynamic overhang speed. Instead of just using bridging fan speed you could define the fan speeds you want for different overhang amounts.

Snuff1eupagus, 2023-02-20:
It would be very nice to have the ability to define a fan speed gradient ramp based on Height range modifiers, or better yet through volume\shape modifiers. We can currently do height range with custom g-code. But it would be nice to have in the form of the GUI.
The Shape based modifiers could simply be placing a volume modifier around an area, the height of the volume is known, and telling it to be X speed at the bottom of the volume, and x speed at the top, you could create multiple volumes to adjust the fan speed very specifically for certain features of your model. perhaps this could work in conjunction with other fan settings, dynamic overhang speed for example.

Godrak, 2023-03-06:
We tested the provided 3mf in the new Alpha 5 version and fan over bridges seems to work correctly.

Also, the upcoming Alpha 5 will introduce Dynamic Fan Speeds in Cooling section. Similarly to dynamic overhang speed, you will be able to set fan speed for certain extrusion overlap levels, with the rest covered by linear interpolation.

closed 2023-03-06 by Godrak

dansity, 2023-03-06:
@Godrak Fan over bridges was never the issue. The issue was fan speed for overhangs not working as it is described in the interface. So either the description is wrong in the interface and this function is non-existent or the function is not working as intended and fan speed is not applied to overhangs (anything less than 90deg).

Godrak, 2023-03-06:
Sorry, I meant to write bridging perimeters. It is currently applied to bridges and also "Overhang Perimeter" features (extrusions marked by dark blue color in Gcode preview in  Feature type mode). Overhang Perimeters are only those that are fully or 'almost' fully unsupported by the previous layer.

The new dynamic overhang speed and upcoming dynamic fan speed should allow you to have much better control over overhangs, if you choose to.