prusa3d/PrusaSlicer #9750: Organic support excessive retract/detract

kris2340k, 2023-02-17:
Can we have an option to just disable retraction for supports

Or at least have supports show as a different object so that I can apply modifiers to it
For example here there are maybe 70-100 retractions and the model has maybe 10. 
https://i.imgur.com/CCAcbfU.jpg

Its completely unnecessary as the supports break off easily regardless and when printing around this you can repeatedly hear retractions every half second.

foreachthing, 2023-02-17:
Already had a jammed nozzle because of that many retractions.

rtyr, 2023-02-17:
Small organic branches would be imho often heavily underextruded after longer nonretracted travel, especially on bowden setups. 

Maybe some "minimal distance for retraction within supports" setting would work universaly for everyone.

kris2340k, 2023-02-17:
Yeah im not asking for mandatory retractions, just an option. It just takes so much time + motor movement vs printing the model.
You can have a 20s layer and 1m40 just retracting on the supports

KrisWragg, 2023-02-18:
Yes just giving this a try myself and this print definitely looks like it is going to be a hot mess of retractions with PETG to support a large flat surface.

[image]

jokohoko, 2023-05-09:
Backing up this one, have been looking for this option and thus came to post a request here.

SphaeroX, 2023-08-15:
I have the same problems, the print works, but the print times are too long for long retracts. Also, I don't think it makes any sense.

Jan-Soustruznik, 2023-08-15:
SPE-1842

SphaeroX, 2023-08-15:
> SPE-1842

What does this mean?

closed 2026-09-29 by prusaslicer-bot[bot]