prusa3d/PrusaSlicer #9752: Slicing error when using multiple tools

nmju99, 2023-02-17:
### Description of the bug

Slicing error when using multiple tools
[image]

### Project file & How to reproduce

[apolloastronaut_supportfree_icz.zip](https://github.com/prusa3d/PrusaSlicer/files/10768327/apolloastronaut_supportfree_icz.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3+win64-202302031521_signed

### Operating system

windows10

### Printer model

toolchanger

lukasmatena, 2023-02-18:
Please, retest in [2.6.0-alpha4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4). if the issue is still there, we will need your project (File->Save as). Thanks.

nmju99, 2023-02-18:
hello！ Organic supports no longer generate contact surfaces, other modes are normal。
[2.6α4.zip](https://github.com/prusa3d/PrusaSlicer/files/10774454/2.6.4.zip)

> Please, retest in [2.6.0-alpha4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha4). if the issue is still there, we will need your project (File->Save as). Thanks.

bubnikv, 2023-03-03:
Only supports with larger overhangs will have soluble interfaces generated, which is not correct. The question is though, whether soluble interfaces at top of thin organic tips will be printable.

[image]

Jan-Soustruznik, 2023-07-14:
SPE-1815

closed 2026-09-29 by prusaslicer-bot[bot]