prusa3d/PrusaSlicer #9753: Slicer freezes on save of G-code

W1SPS, 2023-02-17:
### Description of the bug

Slicer freezes on save of G-code.
Everything up to saving the G-code works fine.

### Project file & How to reproduce

[DPMS-GRIP.zip](https://github.com/prusa3d/PrusaSlicer/files/10769099/DPMS-GRIP.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha3+win64-202302031521_signed

### Operating system

Windows 10 64bit

### Printer model

Prusa MK3s+

neophyl, 2023-02-17:
Once the post processing script link to arcwelder was removed from the profile (as I don't have it so it couldn't find it) saving the gcode took a fraction of a second on my pc.  Windows 10, same version of PS.

So I'd suggest you look at your script as the source of the issue.  At least try it without first.

W1SPS, 2023-02-18:
yup, you were correct. I Thought I had deleted the arc welder line. When I took it out it worked fine.

Thanks,

Rich

Get BlueMail for Android

W1SPS, 2023-02-18:
Althouigh I suppose that is and issue, that ArcWelder doesn't work, or rather the hand off to ArcWelder for processing.

Rich

Get BlueMail for Android

W1SPS, 2023-02-19:
Ok, it's doing it again on Alpha 4 (and 3) every time with no ArcWelder script.

closed 2026-09-29 by prusaslicer-bot[bot]