prusa3d/PrusaSlicer #9756: [FR]  Dynamic overhang speeds on internal perimeters

wuast94, 2023-02-17:
**Is your feature request related to a problem? Please describe.**
Dynamic Overhang Speeds are not  applied to internal perimeters making this feature somehow useless for me. 

**Describe the solution you'd like**
Would be nice to  have the option  to apply these speeds also to internal perimeters

**Describe how it would work**
Just apply the feature to internal perimeters, hope its as easy as i think because the feature is already there 🙂
**Describe alternatives you've considered**
none
**Additional context**

C:\Users\wuast\Downloads\2023-02-17 19_19_21-_Untitled - PrusaSlicer-2.6.0-alpha3 based on Slic3r.png
[3DBench.zip](https://github.com/prusa3d/PrusaSlicer/files/10769920/3DBench.zip)

Printing this benchy makes the front ugly because the 3 internal layers don't get enough cooling and the external perimeter just following the bad internal perimeters

kubispe1, 2023-04-28:
We are looking into it. 
SPE-1648

kubispe1, 2023-05-17:
Hi, with PS 2.6.0 beta1, there is an improvement in dynamic overhang speed, which also affects internal perimeters in this way (Changelog):
 
`The slicer now includes an improved algorithm for overhang speed slowdown and avoiding curled filament. A new and tuned estimation method is applied for the curled height of the filament. The estimated curling now affects the slowdown of overhang speed on external perimeters and also on internal ones`.

I am closing the issue. 
Thanks

closed 2023-05-17 by kubispe1