prusa3d/PrusaSlicer #9757: GitHub releases of pre-release versions (alpha, beta, ... ) marked as latest release version

sroemer, 2023-02-17:
### Description of the bug

GitHub allows to request the latest release version via its API and I am using this in my scripts for automatic installation of a few applications as AppImages on my Linux systems - PrusaSlicer is one of those. When doing so I unexpectedly got an alpha version instead of the expected latest stable release version.

According to the [GitHub API documentation](https://docs.github.com/en/rest/releases/releases?apiVersion=2022-11-28#create-a-release) when creating a release the prerelease value is / should be / will be set to "true to identify the release as a prerelease" and (under make_latest) it says "drafts and prereleases cannot be set as latest".

This would be what I expected to be the case for alpha and beta versions, but currently v2.6.0-alpha3 is the LATEST version and not marked as PreRelease  (see [GitHub API for PrusaSlicer releases](https://api.github.com/repos/prusa3d/PrusaSlicer/releases)). This already to some extend also is mentioned in issue #645, but seems not to be changed.

Obviously this is NOT an important issue and I 'fixed' my script by setting a fix version. Anyway I think the current situation is not ideal and I found it worth pointing that out. Similar to what [quasipedia](https://github.com/quasipedia) said in issue #645, I also never used the release page on my repos myself and therefore cannot help much more. I just digged into the API documentation a bit after I wondered about getting the alpha version I didn't want.

Funny side note:
I installed PrusaSlicer and point out an issue, but not even have any kind of device for using it yet. Just 'played around' with it so far. Seems there is a kind of issue on my side I have to fix as well ;-) . Anyway, thanks for the great work. Keep going!

Kind regards from Vietnam.

### Project file & How to reproduce

Using curl / jq to request latest version:
`curl -sL https://api.github.com/repos/prusa3d/PrusaSlicer/releases/latest | jq -reM ".tag_name"`

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3 and previous alpha / beta versions

### Operating system

does not apply

### Printer model

does not apply

lukasmatena, 2023-02-17:
@sroemer Thanks. I can see that I indeed marked 2.6.0-alpha3 as the latest release (it probably was the default). I will try not to repeat this mistake again. I changed the latest to 2.5.0.

It has little to do with #645, that was about non-standard versioning - we are now using semver for quite a long time. The latest/prerelease are github-specific flags.

Closing.

closed 2023-02-17 by lukasmatena

sroemer, 2023-02-17:
@lukasmatena Great, it works. Thanks, I didn't expect such a quick change here.

Regarding #645: I know the main purpose of that issue was the versioning scheme itself, but when you read through it you will find for example the following part which is to some extend related:

> The main thing is though: if you are using semver (not perfect, but by far the most common versioning scheme and the one GitHub comes with "magic" out of the box), beta iterations substitute patch ones, as they serve the same purpose (fixing bugs). The thing is: an alpha or beta or RC in semver is not considered a proper "release" but rather a "prerelease". This is why projects that stick to it often offer two or three "channels" (regular releases, pre-releases, dev snapshots).

And if I understand your comment correctly, then you currently do have to select between Pre-Release or Release manually and/or set the latest version manually. Even if a mistake in that case is not a big deal, doing it like this is error prone. My understanding of the quote above was that there is a somehow automatic way of making a version a pre-release if they are named alpha, beta, or so, but as I said before I don't know much about the GitHub release process/API. Maybe I will dig a bit deeper into it soon ...