prusa3d/PrusaSlicer #9758: Add a perimeters overlap setting

BMacours, 2023-02-17:
**Is your feature request related to a problem? Please describe.**
Perimeters not wanting to bind together is an occuring and highly annoying issue when printing small technical parts (in my case with a Prusa MINI, a 0.25 nozzle and high quality filament).

**Describe the solution you'd like**
Having a feature allowing to specify overlap (mm or %) between perimeters would be an easy, elegant and effective way to solve the issue.
As described in PrusaSlicer help https://help.prusa3d.com/article/layers-and-perimeters_1748#recommended-thin-wall-thickness : "PrusaSlicer slightly overlaps the perimeters". This "slightly" would be welcome a slicer overlap setting!

**Describe how it would work**
Add this setting in PrusaSlicer/Print Settings/Advanced/Overlap.

**Describe alternatives you've considered**
Experimenting to change extrusion multiplier and/or extrusion width sometimes do the job after time consuming experiments but causes potential XY dimentional accuracy losses which is a no go for small technical parts.

Note that I already tested everything (i think) I could do : belt adjustment, nozzle change, print slower, hotter, ... but there is always a corner case that my Mini can't handle properly.

BMacours, 2023-02-20:
After looking into PrusaSlicer C++ code and making some experimental changes (I'm no C++ expert) I found a trick to make internal perimeters to overlap themselves a bit more. This solves the "perimeters not bonding" issue at leat in my case.

Of course I know you know what the following piece of code does and what are the many potential side effects (gap fill, overhangs, ...). This is maybe not the way to go but I'd like to share and demonstrate how this kind of tuning can potentially solve the "not bonding perimeters" issue.

So I added a 0.85 factor to the offset distance to increase perimeter overlap :
`PerimeterGenerator.cpp (2.6 alpha 4)`
`line 1437 : offset_ex(last, - float(distance) * 0.85 /* Increase perimeter overlap by 15% */);`

Cheers,
Bernard

BMacours, 2023-02-25:
I'm responding to myself haha.

So I investigated further and think I found a more elegant way in PerimeterGenerator.cpp

`...`
`#define EXTRA_PERIMETER_OVERLAP 0.20 // Additional perimeter overlap (percent) => Should become a dynamic setting`
`#define EXTRA_EXTERNAL_PERIMETER_OVERLAP 0.20 // Additional external perimeter overlap (percent) => Should become a dynamic setting`
`...`
`// Adjust perimeter offset distance in process_classic method (betwee line 1420 et 1430)`
`coord_t distance = (i == 1) ? coord_t(ext_perimeter_spacing2 * (1.00 - EXTRA_EXTERNAL_PERIMETER_OVERLAP)) : coord_t(perimeter_spacing * (1.00 - EXTRA_PERIMETER_OVERLAP));`
`...`

For the small technical pieces I'm printing, this definitely solves the "not bonding perimeters issue" and I honestly didn't spot any drawbacks but slight overextrusions at some places. But I highly prefer to deal with some overextrusions than with not bonding perimeters which ruins my prints.

bubnikv, 2023-03-03:
If the perimeters are not binding together, then you need to correctly
calibrate your extruder flow and / or linear advance.

BMacours, 2023-03-17:
Hi, thank you for you comment.

Indeed this solves the issue (I tried of course) but it has an impact on all extrusion types (infills) while I only have some perimeters at some places that do not bind. Moreover increasing flow ratio decreases dimentional accuracy. If wouldn't car if I was printing non-technical parts where every 0.1 mm counts.

3D printing still has some lack of precision due to vibrations and not perfectly square axis (especially XY with the Mini). IMHO an additional slicing parameter to easily overcome this perimeter issue would be welcome IMHO.

BMacours, 2026-09-22:
Still in my wish list. Thx.

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16123.

closed 2026-10-21 by Jan-Soustruznik