prusa3d/PrusaSlicer #9759: AnkerMaker M5

redfoxonhigh, 2023-02-17:
### Description of the bug

The time displayed started out at 224122 hours and 23 minutes. Then it jumped higher before going to 1 hour 32 minutes. It was 
 only about a half-inch tall. It bounces from 32 minutes to 1 hour 34 minutes. No idea how long it is going to take. I side it is running at 6x speed.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha3

### Operating system

windows 11

### Printer model

AnkerMaker M5

redfoxonhigh, 2023-02-17:
Please let me know when there is an update.
Thanks

rtyr, 2023-02-18:
I am afraid it is up to AnkerMake to fix. FW likely doesn't expect g-code from 3rd party slicers.

closed 2026-09-29 by prusaslicer-bot[bot]