prusa3d/PrusaSlicer #9765: Seam Painting (Fix/add new feature)

jasonhartsoe, 2023-02-18:
**Is your feature request related to a problem? Please describe.**
Yes.  When using "Seam painting" it never creates a straight seam.  It generates a random all over the place uneven non straight line when you paint on the seam.  Regardless what method or setting you use, it will never seem to get that perfect straight line generated.  There are some ways to get it to work, but this has continued to be difficult in most complex models.  

**Describe the solution you'd like**
I'd like to see a few things happen.  One, fix the current paint on seam....When you are painting, it should be able to take the direction and based on space, know that there's room to create a 90 degree or straight line in the direction painted.  So If I paint a line straight down, it should create a perfectly straight 90 degree perpendicular line straight down for the seam and not sideways, or randomly sidewinding all the way down.  Just a perfectly straight line in the direction and placement you painted.

The additional option I would like to see added is the option to select a line as if I was drawing in fusion 360.  

**Describe how it would work**
For the new additional option, instead of selecting a circle or sphere to draw with, you can select a "line".  You can then select the starting point and end point so it creates a perfectly straight line from top to bottom and creates the perfectly straight seam in that direction.  I'd leave it open to any angle for the user to select, but have the option to either snap to a perfect 90 degree straight down from top to bottom option and/or also add a shift button option so after you selected your first line, you then press and hold the shift button (or you can do this from the start) and it will create the line in 45 degree increments.  So it will give you the perfectly straight top to bottom line, but also if you move left or right a 45 degree change etc.  So one, the user can create any angle with it, or hold the shift button down to get the perfectly straight line straight down.  It would be nice to get that perfect seam every single time without having to struggle painting over and over and over.

**Describe alternatives you've considered**
No.

**Additional context**
Thanks!

ipsod, 2023-04-01:
I came here looking for something like this.

I just want a seam brush that's a long vertical line.  I've got mechanical parts that need seams to hide in particular places, and it's difficult with the circle/sphere brush.

lukasmatena, 2024-02-19:
Closing as a duplicate of #5970.

closed 2024-02-19 by lukasmatena