prusa3d/PrusaSlicer #9768: Need different fixes in interface

AndylgTom, 2023-02-18:
### Description of the bug

I'll just throw screenshots here, it should be clear what needs to be fixed.

### Project file & How to reproduce

[image]

[image]

[image]

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0 alpha4

### Operating system

Win 10

### Printer model

never mind

AndylgTom, 2023-03-07:
In version PrusaSlicer 2.6.0-alpha5 only fixes problem #2... The other problems remain.

YuSanka, 2023-03-08:
1. and 3. issues are caused by new wxWidgets, we will fix it in some of next releases. But I recommend you to use system menu for application, than you wouldn't have those issues. 
For enabling system menu check "Use system menu for application" in "Preferences".

lukasmatena, 2023-04-02:
Thanks. All the issues should be fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena