prusa3d/PrusaSlicer #9770: filament type not user editable, dropdown only

KrisWragg, 2023-02-18:
### Description of the bug

Using 2.6.0-alpha4 my PETG filament profile had an empty filament type, it seems to have change between 2.5.0 and the 2.6.0 in that it is no longer user editable and can only contain items in the list, of which PETG is not an option.

This breaks some of of my printer settings that were checking whether type was PETG to apply a z offset. I can work around this for now but it would be nice to either get PETG in the list or make it user editable again like 2.5.0

### Project file & How to reproduce

Filament Settings -> Advanced -> Filament type

[image]

Example ini file from 2.5.0 attached below

[Overture Black PETG ini.zip](https://github.com/prusa3d/PrusaSlicer/files/10774520/Overture.Black.PETG.ini.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Windows 11

### Printer model

Ender 3 V2

TheWebMachine, 2023-02-22:
I can concur that this is happening only in 2.6.0-alpha4 and was NOT a bug in alpha3. It is also a breaking change for me, as I rely on custom filament types in code (like different PLAs; PLA+, Meta, metal-fill, etc).

mfraser, 2023-03-01:
For me the filament type doesn't change from PLA even when selecting PETG, but if you compare a PLA & PETG filament it shows that they are different.

plampix, 2023-03-08:
Thanks for marking my issue as a duplicate, I couldn't find anything directly referencing the PETG->PET change.

lukasmatena, 2023-04-02:
Fixed in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena