prusa3d/PrusaSlicer #9771: Connectors/cutting operations

jtessens, 2023-02-18:
### Description of the bug

[bug after exporing build plate.zip](https://github.com/prusa3d/PrusaSlicer/files/10774703/bug.after.exporing.build.plate.zip)

After exporting build plate the plugs are gone, the dowels remain.

During cutting/connector operations: When changing from dowels to plugs (even after removing them) there are dowels on both objects.

### Project file & How to reproduce

Edit connectors, remove dowels, change to plugs.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 4

### Operating system

Windows 10

### Printer model

Ender 3 V2

YuSanka, 2023-02-20:
Thanks for your report. But I can't understand your issue. 
Cut information in 3mf you provide us looks uncompleted.
Could you please provide us actual steps to reproduce the issue.

jtessens, 2023-02-20:
Dear,

I load model.
I go to cutting tool.
I make (edit) the cutting pane
I add the connectors.
Confirm the connectors.
Perform the cut.
In the sliced model the cuts and connectors are as supposed to be.
I export build plate.
I close PrusaSlicer.
I reopen the exported model.
The dowels are there...the plugs are gone. (That's the file I send).

2nd problem
.... Same steps as above. (Fresh opened file)
I add the connectors.
I want to change dowel to plug connector.
I press remove connectors.
I select the plug option and place the connector.
Instead of plug I get a dowel...on both sides.

Kind regards,
Johan Tessens

YuSanka, 2023-02-20:
OK, dear )))

Regarding to the first problem, you exported build plate to the STL. 
That is why all information related to the cut is gone. See #7165.

But second part of issue we still cannot to reproduce. May be you could provide some screenshots?

YuSanka, 2023-02-21:
@jtessens , There is no attachment. 
You have to upload it using the web interface, email attachments do not work. 
Thanks.

jtessens, 2023-02-21:
[screen captures.pdf](https://github.com/prusa3d/PrusaSlicer/files/10792111/screen.captures.pdf)

YuSanka, 2023-02-21:
OK, I finally understand your problem. 
Probably accidently, but you flipped the cut plane upside down during manipulation with cut plane. 
That is why plugs are put to the upper part.
Note the color of the part where you put connectors. Purple part is "base" part for the plugs.
Does it solve your problem?

jtessens, 2023-02-21:
I attached another series of screenshots with explanation how I proceed.
[cutting and connectors.pdf](https://github.com/prusa3d/PrusaSlicer/files/10793778/cutting.and.connectors.pdf)

lukasmatena, 2023-02-21:
@jtessens Thanks. We probably see what you mean and where the problem is. There are two cutting modes:
- Plug: one part is a male, the other is female
- Dowel: both part have holes, they will be connected by a dowel (I'd rather not use the gender analogy here for quite obvious reasons...)

Therefore, the outcome is correct. It is correct that both your parts have holes and two extra rods (dowels) emerged on the print bed. The only misleading part is the preview of the holes when adding them. They protrude quite a bit above the cutting plane, making an impression that a pin has been placed. In reality, the cylinder that is shown is a negative volume, so it really is a hole. We will shorten the negative mesh so it just barely looks out of the surface to make it more clear.

Do we understand each other now? Do you think that this change would make the tool easier to understand?

kubispe1, 2023-03-01:
Visual changes were made and in the next release 2.6.0-alfa5, negative volumes are shortened for dowels and for plug holes. 
We tested it according to SPE-1512. I am closing the issue. Hope it will be much more clear to use. 
Closing the issue

closed 2023-03-01 by kubispe1