prusa3d/PrusaSlicer #9772: Organic support created in mid air

engofthings, 2023-02-18:
### Description of the bug

Organic support was generated, a branch started inside  **not from the bottom layer,** but on a higher layer, creating extrusion in mid air. The tree should not include extrusions in mid air.
For sequential Z heights, starting from Z=2

[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Bulbasaur - bad organic support genereation.zip](https://github.com/prusa3d/PrusaSlicer/files/10775010/Bulbasaur.-.bad.organic.support.genereation.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha3+win64

### Operating system

Windows 10

### Printer model

Mk3S+

russphelan, 2023-02-24:
I am also having this issue: 
[image]

Here is some system and version info: 
[image]

Please let me know if there's something else useful I can do, thanks.

kubispe1, 2023-02-24:
Thanks for the issue and painted Bulbasaur. I will try to print it next week.
This issue is a duplicate. #9601 
I am closing this issue.

closed 2023-02-24 by kubispe1