prusa3d/PrusaSlicer #9774: Crash on start - SIGSEGV in ImFont::CalcTextSizeA

mkyral, 2023-02-18:
### Description of the bug

PrusaSlicer 2.6.0-alpha4 compiled from sources on my Gentoo  is crashing on start. The pre-compilled binary downloaded from github works well. I've tested start with clean profile, but still crashing.

### Backtrace:
```
Reading symbols from /usr/bin/prusa-slicer...
Reading symbols from /usr/lib/debug//usr/bin/prusa-slicer.debug...
(gdb) run
Starting program: /usr/bin/prusa-slicer
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[2023-02-18 20:45:04.884193] [0x00007fffeebfe9c0] [trace]   Initializing StaticPrintConfigs
[Detaching after fork from child process 28436]
[New Thread 0x7fffee53f6c0 (LWP 28437)]
[New Thread 0x7fffedd3e6c0 (LWP 28438)]
[New Thread 0x7fffed53d6c0 (LWP 28439)]
[New Thread 0x7fffecd3c6c0 (LWP 28440)]
[New Thread 0x7fffdffff6c0 (LWP 28441)]
[Thread 0x7fffdffff6c0 (LWP 28441) exited]
[New Thread 0x7fffdffff6c0 (LWP 28442)]
[New Thread 0x7fffdf13c6c0 (LWP 28443)]
[Thread 0x7fffdffff6c0 (LWP 28442) exited]
[Thread 0x7fffdf13c6c0 (LWP 28443) exited]
[New Thread 0x7fffdf13c6c0 (LWP 28444)]
[New Thread 0x7fffdffff6c0 (LWP 28445)]
[Thread 0x7fffdf13c6c0 (LWP 28444) exited]
[New Thread 0x7fffdf13c6c0 (LWP 28446)]
[New Thread 0x7fffde93b6c0 (LWP 28447)]
[Thread 0x7fffdf13c6c0 (LWP 28446) exited]
[Thread 0x7fffdffff6c0 (LWP 28445) exited]
[New Thread 0x7fffdffff6c0 (LWP 28448)]
[New Thread 0x7fffdf13c6c0 (LWP 28449)]
[Thread 0x7fffde93b6c0 (LWP 28447) exited]
[Thread 0x7fffdffff6c0 (LWP 28448) exited]
[Thread 0x7fffdf13c6c0 (LWP 28449) exited]
[New Thread 0x7fffdf13c6c0 (LWP 28450)]
[New Thread 0x7fffdffff6c0 (LWP 28451)]
[Thread 0x7fffdf13c6c0 (LWP 28450) exited]
[Thread 0x7fffdffff6c0 (LWP 28451) exited]
[New Thread 0x7fffdffff6c0 (LWP 28452)]
[New Thread 0x7fffdf13c6c0 (LWP 28453)]
[Thread 0x7fffdffff6c0 (LWP 28452) exited]
[Thread 0x7fffdf13c6c0 (LWP 28453) exited]
[New Thread 0x7fffdf13c6c0 (LWP 28454)]
[New Thread 0x7fffdffff6c0 (LWP 28455)]
[Thread 0x7fffdf13c6c0 (LWP 28454) exited]
[Thread 0x7fffdffff6c0 (LWP 28455) exited]
[New Thread 0x7fffdffff6c0 (LWP 28456)]
[New Thread 0x7fffdf13c6c0 (LWP 28457)]
[New Thread 0x7fffde93b6c0 (LWP 28458)]
[New Thread 0x7fffdcdff6c0 (LWP 28477)]
[New Thread 0x7fffc97ab6c0 (LWP 28478)]
[New Thread 0x7fffc8faa6c0 (LWP 28479)]
[2023-02-18 20:45:05.824344] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[2023-02-18 20:45:05.825244] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[2023-02-18 20:45:05.858521] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[2023-02-18 20:45:05.859554] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[Thread 0x7fffde93b6c0 (LWP 28458) exited]
[Thread 0x7fffecd3c6c0 (LWP 28440) exited]

Thread 1 "slic3r_main" received signal SIGSEGV, Segmentation fault.
0x000055555693b9b5 in ImFont::CalcTextSizeA (this=0x0, size=0, max_width=max_width@entry=3.40282347e+38, wrap_width=-1, text_begin=text_begin@entry=0x555556a406fc "A", text_end=0x555556a406fd "", remaining=remaining@entry=0x0) at /usr/src/debug/media-gfx/prusaslicer-2.6.0_alpha4/PrusaSlicer-version_2.6.0-alpha4/src/imgui/imgui_draw.cpp:3408
3408        const float scale = size / FontSize;
(gdb) bt
#0  0x000055555693b9b5 in ImFont::CalcTextSizeA(float, float, float, char const*, char const*, char const**) const
    (this=0x0, size=0, max_width=max_width@entry=3.40282347e+38, wrap_width=-1, text_begin=text_begin@entry=0x555556a406fc "A", text_end=0x555556a406fd "", remaining=remaining@entry=0x0)
    at /usr/src/debug/media-gfx/prusaslicer-2.6.0_alpha4/PrusaSlicer-version_2.6.0-alpha4/src/imgui/imgui_draw.cpp:3408
#1  0x00005555568f6a38 in ImGui::CalcTextSize(char const*, char const*, bool, float)
    (text=text@entry=0x555556a406fc "A", text_end=text_end@entry=0x0, hide_text_after_double_hash=hide_text_after_double_hash@entry=false, wrap_width=wrap_width@entry=-1)
[... 153 lines cut ...]
-- Generating done
-- Build files have been written to: /var/tmp/portage/media-gfx/prusaslicer-2.6.0_alpha4/work/PrusaSlicer-version_2.6.0-alpha4_build
>>> Source configured.
```

### Project file & How to reproduce

1) Run prusa-slicer ;-)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Gentoo Linux

### Printer model

MK3S

lukasmatena, 2023-02-19:
Seems to be related to #9780.

lukasmatena, 2023-02-19:
The backtrace would indicate that the call to `CalcTextSize` happens before imgui font is initialized. Same problem is described in https://github.com/ocornut/imgui/issues/4769.

kocikdav, 2023-02-20:
Hello, for those who compile from git, there is a branch dk_hints_public with commit 6cb99470664d4d493992c4fe2925aa43268c7a4e. Hopefully this fixes the issue. For the rest, Im going to prepare a custom build and share it under this issue.

kocikdav, 2023-02-21:
Previous url might not work due security. These should:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-linux-x64-GTK3-g6cb9947-202302201436.AppImage
 https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-MacOS-universal-g6cb994706-202302201434.dmg
or enter
user: prusaslicer
password: slicer
if dialog shows.

mkyral, 2023-02-26:
Hi,
I've tested the mentioned commit on alpha4.
No error on start, but the 3D view is not initialized. Other tabs are working, but not 3D view.

In console I can see many errors with glew initialization. 
```
[2023-02-26 08:48:43.633681] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[2023-02-26 08:48:43.634523] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[2023-02-26 08:48:44.071529] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
[2023-02-26 08:48:44.072370] [0x00007fffeebfe9c0] [error]   Unable to init glew library: Unknown error
```
This is not an issue with pre-compiled binaries. Probably because the pre-compiled binary is not linked to glew library.
```
[08:54:45 marian@txp bin]$ ./prusa-slicer -h |grep PrusaSlicer-2
PrusaSlicer-2.6.0-alpha4+linux-x64-GTK3-202302180033 based on Slic3r (with GUI support)
[08:55:19 marian@txp bin]$ ldd prusa-slicer |grep -i glew
[08:55:19 marian@txp bin]$
```
```
[08:55:45 marian@txp bin]$ /usr/bin/prusa-slicer -h |grep PrusaSlicer-2
PrusaSlicer-2.6.0-alpha4+UNKNOWN based on Slic3r (with GUI support)
[08:55:58 marian@txp bin]$ ldd /usr/bin/prusa-slicer |grep -i glew
        libGLEW.so.2.2 => /usr/lib64/libGLEW.so.2.2 (0x00007f3c0b8f3000)
```
[image]

mkyral, 2023-02-26:
Probably it could be this: https://github.com/wxWidgets/wxWidgets/issues/22710

I'm not sure what exactly I should do when in Gentoo I have x11-libs/wxGTK-3.2.2.1 installed.

lukasmatena, 2023-03-06:
@mkyral Could you please test this with [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5)? Thanks.

mkyral, 2023-03-18:
Hi @lukasmatena.
sorry for delay. I can confirm that alpha-5 works with distribution wxGTK 3.2

slashbeast, 2023-04-01:
Locally built alpha6 with wxGTK 3.2 and glew 2.2.0 it fails the same for me. Since now you use `wxBitmapBundle` the wx 3.2 is required and it no longer can be built with wx 3.0.

```
% prusa-slicer 
[2023-04-01 18:06:57.655410] [0x00007f47678e0a00] [trace]   Initializing StaticPrintConfigs

(prusa-slicer:23969): Gtk-CRITICAL **: 18:06:58.083: gtk_distribute_natural_allocation: assertion 'extra_space >= 0' failed
[2023-04-01 18:06:58.130270] [0x00007f47678e0a00] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:06:58.130881] [0x00007f47678e0a00] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:06:58.132555] [0x00007f47678e0a00] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:06:58.133142] [0x00007f47678e0a00] [error]   Unable to init glew library: Unknown error
zsh: segmentation fault  prusa-slicer

```

The appimage `PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518.AppImage` does work just fine. 

However because you do not publish any build scripts how those AppImage images are put together I have no way to find what you do there to make it work. Would be really great if you could share those.

slashbeast, 2023-04-01:
@lukasmatena I followed with your commend regarding show_hints from https://github.com/prusa3d/PrusaSlicer/issues/9780 and if I modify PrusaSlicer.ini to have show_hints set to 0, locally built 2.6.0_alpha6 does start up, however it does not render at all the Plater tab though I can interact with UI elements and switch tabs.

```
% prusa-slicer 
[2023-04-01 18:44:56.291627] [0x00007fb8c52a6a80] [trace]   Initializing StaticPrintConfigs

(prusa-slicer:9052): Gtk-CRITICAL **: 18:44:57.555: gtk_distribute_natural_allocation: assertion 'extra_space >= 0' failed
[2023-04-01 18:44:57.605352] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.605961] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.607729] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.608312] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.658343] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.658934] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.670567] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:57.671156] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:58.038187] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error
[2023-04-01 18:44:58.038770] [0x00007fb8c52a6a80] [error]   Unable to init glew library: Unknown error

(prusa-slicer:9052): Gtk-CRITICAL **: 18:44:58.434: gtk_widget_event: assertion 'WIDGET_REALIZED_FOR_EVENT (widget, event)' failed
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it
18:44:58: Debug: window wxTreeCtrl@0x5555b1fc8c40 ("treeCtrl") lost focus even though it didn't have it

```

mkyral, 2023-05-14:
Just tested 2.6.0-beta1 and issue remains. No 3D view. I have to recompile wxGTK with `--disable-glcanvasegl` option.

slashbeast, 2023-07-23:
Following the example of Fedora[1] I will force the GLCanvas GLX in Gentoo along with PrusaSlicer 2.6.0. I can confirm that it works in full, though still requires a handful of patches to make it build, and the wxWidgets new asserts are noisy as well, patched it out.

Perhaps the README of repo should state to disable the GLCanvas EGL as requirement and then this issue could be closed.

[1] https://src.fedoraproject.org/rpms/wxGTK/c/4b153842efce56595f3320fa9d888741060a2464

lukasmatena, 2023-07-24:
> Perhaps the README of repo should state to disable the GLCanvas EGL as requirement and then this issue could be closed.

Thanks. In the next release, this information will be added into `doc/How to build - Linux et al.md`, along with a reference to this issue.
Closing.

closed 2023-07-24 by lukasmatena