prusa3d/PrusaSlicer #9777: Unable to Merge after using Split

sliverbaer, 2023-02-18:
### Description of the bug

**Issue:**
After using the split command, unable to merge the parts.  

**Expectation:**
Should be able to split part move the parts around and then merge.

### Project file & How to reproduce

**Steps:**
Go to part.  Split into multiple parts.  Maybe copy one to make the part longer, align them up.  Select all and then right click.  No option to merge.  (Unable to attach 3mf file.)

**Workaround:**
If you copy all the parts and then paste them all.  Then select the ones you just pasted, you then get the option to merge.

[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4+win64

### Operating system

WIndows 10

### Printer model

Prusa i3 MK3S+ & SliceWorx KP3S Pro S1

neophyl, 2023-02-19:
You need to zip up your 3mf file to attach it.

sliverbaer, 2023-02-19:
I tried this again in 2.6a4 and it wasn't giving the same icon as the 2.6a3.  I tried with another model just to make sure in both 2.6a3 and 2.6a4.  It was only doing this in 2.6a3 with a new (different) model.

It seems it was resolved in 2.6a4.  I will go ahead and close out.

closed 2023-02-19 by sliverbaer