prusa3d/PrusaSlicer #9778: cant figure out where to update prusaslicer 2.6.0 alpha 4

NVIXTY, 2023-02-19:
### Description of the bug

i dont know where to press to put the .zip file into prusaslicer

### Project file & How to reproduce

its the .zip file i cant put it here cuz its too big

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 10

### Printer model

ender 3 pro with some mods

foreachthing, 2023-02-19:
Just extract the zip file anywhere you like. No need to replace anything - you'd be better off not replacing your current install of 2.5.
Then make a copy of your [profiles ](https://projects.ttlexceeded.com/3dprinting_prusaslicer_profiles.html) (or PrusaSlicer -> Help -> Show Configuration Folder) and use the "[-datadir "path"](https://github.com/prusa3d/PrusaSlicer/issues/5023)" to make sure you don't mess up your non-alpha profiles.

Example shortcut: `"c:\PrusaSlicer2.6\prusaslicer.exe" -datadir "c:\prusaslicer_profiles\profiles26a"`

bubnikv, 2023-03-03:
answered, closing

closed 2023-03-03 by bubnikv