prusa3d/PrusaSlicer #9779: PrusaSlicer2.6alpha4 Cannot set locale to language "Chinese (Simplified, China)".

wisonlau, 2023-02-19:
### Description of the bug

[image]

```log
MacOS ./PrusaSlicer --loglevel=5
[2023-02-19 11:11:27.943257] [0x0000000113108600] [trace]   Initializing StaticPrintConfigs
[2023-02-19 11:11:27.952650] [0x0000000113108600] [trace]   App updater default download path: "/Users/wisonlau/Downloads"
[2023-02-19 11:11:27.952725] [0x0000000113108600] [debug]   full path: 4072512567279585534
[2023-02-19 11:11:27.952736] [0x0000000113108600] [debug]   single instance: undefined. other params: ./PrusaSlicer;--loglevel=5
[2023-02-19 11:11:27.952742] [0x0000000113108600] [debug]   full lock path: /Users/wisonlau/Library/Application Support/PrusaSlicer-alpha/cache/4072512567279585534.lock
[2023-02-19 11:11:27.954183] [0x0000000113108600] [debug]   Creating lockfile.
[2023-02-19 11:11:27.954213] [0x0000000113108600] [info]    Instance check: Another instance not found or single-instance not set.
[2023-02-19 11:11:28.057299] [0x0000000113108600] [trace]   translation_language provided by PrusaSlicer.ini: zh_CN
[2023-02-19 11:11:28.060055] [0x0000000113108600] [trace]   System language detected (user locales and such): zh_CN
[2023-02-19 11:11:28.063184] [0x0000000113108600] [trace]   Best translation language detected (may be different from user locales): zh_TW
[2023-02-19 11:11:28.063438] [0x0000000113108600] [trace]   Switching wxLocales to zh_CN
[2023-02-19 11:11:28.198854] [0x0000000113108600] [info]    Checking if indices need to be installed from resources...
[2023-02-19 11:11:28.638696] [0x0000000113108600] [info]    3dx mac handler starts
[2023-02-19 11:11:28.638877] [0x0000000113108600] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file)
[2023-02-19 11:11:28.638887] [0x0000000113108600] [info]    3dx drivers loaded: NO
[2023-02-19 11:11:28.641071] [0x0000000113108600] [info]    Undo / Redo snapshot taken: 新项目, Undo / Redo stack memory: 0MB Resident memory: 128MB; Peak memory usage: 128MB
[2023-02-19 11:11:28.731605] [0x0000000113108600] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2023-02-19 11:11:28.801485] [0x0000000113108600] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2023-02-19 11:11:28.801524] [0x0000000113108600] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2023-02-19 11:11:29.440854] [0x0000000113108600] [debug]   TriangleMesh::repair() started
[2023-02-19 11:11:29.441722] [0x0000000113108600] [debug]   TriangleMesh::repair() finished
[2023-02-19 11:11:30.844670] [0x0000000113108600] [error]   Hint notification without hypertext end marker: Hiding sidebar\nDid you know that you can hide the right sidebar using the shortcut <b>Shift+Tab</b>? You can also enable the icon for this from the<a>Preferences</a>.
[2023-02-19 11:11:30.844948] [0x0000000113108600] [error]   Hint notification without hypertext end marker: Configuration snapshots\nDid you know that you can roll back to a complete backup of all system and user profiles? You can view and move back and forth between snapshots using the Configuration - <a>Configuration snapshots menu</a>.
[2023-02-19 11:11:30.844989] [0x0000000113108600] [error]   Hint notification without hypertext end marker: Minimum shell thickness\nDid you know that instead of the number of top and bottom layers, you can define the<a>Minimum shell thickness</a>in millimeters? This feature is especially useful when using the variable layer height function.
[2023-02-19 11:11:30.847331] [0x0000000113108600] [info]    Checking for cached configuration updates...
[2023-02-19 11:11:30.847380] [0x0000000113108600] [info]    Confing bundle not installed for vendor Rigid3D, skipping:
[2023-02-19 11:11:30.847404] [0x0000000113108600] [info]    Confing bundle not installed for vendor TriLAB, skipping:
[2023-02-19 11:11:30.847425] [0x0000000113108600] [info]    Confing bundle not installed for vendor Geeetech, skipping:
[2023-02-19 11:11:30.847440] [0x0000000113108600] [info]    Confing bundle not installed for vendor Anker, skipping:
[2023-02-19 11:11:30.847451] [0x0000000113108600] [info]    Confing bundle not installed for vendor Print4Taste, skipping:
[2023-02-19 11:11:30.847462] [0x0000000113108600] [info]    Confing bundle not installed for vendor Elegoo, skipping:
[2023-02-19 11:11:30.847473] [0x0000000113108600] [info]    Confing bundle not installed for vendor Snapmaker, skipping:
[2023-02-19 11:11:30.851416] [0x0000000113108600] [debug]   Vendor: Templates, version installed: 1.0.0, version cached: 1.0.0
[2023-02-19 11:11:30.851475] [0x0000000113108600] [info]    Confing bundle not installed for vendor BIBO, skipping:
[2023-02-19 11:11:30.851491] [0x0000000113108600] [info]    Confing bundle not installed for vendor LulzBot, skipping:
[2023-02-19 11:11:30.867773] [0x0000000113108600] [debug]   Vendor: Prusa Research, version installed: 1.6.0-alpha1, version cached: 1.6.0-alpha1
[2023-02-19 11:11:30.867805] [0x0000000113108600] [info]    Confing bundle not installed for vendor Ultimaker, skipping:
[2023-02-19 11:11:30.867814] [0x0000000113108600] [info]    Confing bundle not installed for vendor Voron, skipping:
[2023-02-19 11:11:30.867821] [0x0000000113108600] [info]    Confing bundle not installed for vendor Infinity3D, skipping:
[2023-02-19 11:11:30.867829] [0x0000000113108600] [info]    Confing bundle not installed for vendor Jubilee, skipping:
[2023-02-19 11:11:30.867835] [0x0000000113108600] [info]    Confing bundle not installed for vendor FLSun, skipping:
[2023-02-19 11:11:30.867842] [0x0000000113108600] [info]    Confing bundle not installed for vendor Zonestar, skipping:
[2023-02-19 11:11:30.867848] [0x0000000113108600] [info]    Confing bundle not installed for vendor Anycubic, skipping:
[2023-02-19 11:11:30.867855] [0x0000000113108600] [info]    Confing bundle not installed for vendor gCreate, skipping:
[2023-02-19 11:11:30.867861] [0x0000000113108600] [info]    Confing bundle not installed for vendor Creality, skipping:
[2023-02-19 11:11:30.867868] [0x0000000113108600] [info]    Confing bundle not installed for vendor RatRig, skipping:
[2023-02-19 11:11:30.867874] [0x0000000113108600] [info]    Confing bundle not installed for vendor BIQU, skipping:
[2023-02-19 11:11:30.867881] [0x0000000113108600] [info]    Confing bundle not installed for vendor Artillery, skipping:
[2023-02-19 11:11:30.867888] [0x0000000113108600] [info]    Confing bundle not installed for vendor INAT, skipping:
[2023-02-19 11:11:30.867894] [0x0000000113108600] [info]    Confing bundle not installed for vendor MakerGear, skipping:
[2023-02-19 11:11:30.867905] [0x0000000113108600] [info]    No configuration updates available.
[2023-02-19 11:11:31.901580] [0x0000700004d05000] [info]    Syncing configuration cache
[2023-02-19 11:11:31.901623] [0x0000700004d05000] [info]    Downloading vedor profiles archive zip from https://files.prusa3d.com/wp-content/uploads/repository/vendor_indices.zip
[2023-02-19 11:11:31.901667] [0x0000700004d05000] [info]    Get: `https://files.prusa3d.com/wp-content/uploads/repository/vendor_indices.zip`
	-> `/Users/wisonlau/Library/Application Support/PrusaSlicer-alpha/cache/vendor_indices.zip`
	via tmp path `/Users/wisonlau/Library/Application Support/PrusaSlicer-alpha/cache/vendor_indices.zip.20301.download`
*   Trying 185.115.1.124:443...
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
[... 140 lines cut ...]
[2023-02-19 11:11:33.027467] [0x0000700004d05000] [info]    Resource PrusaResearch / sl1s_bed.stl found in resources folder. No need to download.
[2023-02-19 11:11:33.027484] [0x0000700004d05000] [info]    Resource PrusaResearch / SL1S_thumbnail.png found in resources folder. No need to download.
[2023-02-19 11:12:13.245498] [0x0000000113108600] [debug]   message handler shutdown().
```

### Project file & How to reproduce

https://github.com/prusa3d/PrusaSlicer/issues/9476

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha4+MacOS-universal-202302180021.dmg

### Operating system

macos 12.6.1

### Printer model

prusa mk3s

ycbwb, 2023-03-10:
The same problem.
mac os 13.2.1
version: alpha5

AlexTOOT, 2023-06-03:
Same problem
2.6.0 Alpha6
Macos 13.2.1

gongzuo51, 2023-06-04:
Same problem
2.6.0 b4
macos 13.4

lukasmatena, 2023-06-09:
Thanks for the report. Fixed in [2.6.0-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc1). Closing.

closed 2023-06-09 by lukasmatena