prusa3d/PrusaSlicer #9780: PrusaSlicer 2.6.0-Alpha4 crashes on start on OS X Ventura 13.2.1

kb9ysc, 2023-02-19:
### Description of the bug

Prusaslicer 2.6.0 Alpha 4 (and 3) crashes on startup.

Expected behavior is that the program starts up.

Appears to have started after upgrade to Ventura 13.2.1.

### Project file & How to reproduce

When I try to start PrusaSlicer 2.6.0-Alpha4,  it crashes upon load.  I just updated to OS X 13.2.1, and that's when the issue started.  I saw the issue with Alpha3 but Alpha 4 is doing the same.  I even tried to download the sources and compile locally,  it was still doing the same.  Then downloaded Alpha4 when I noticed it was available.

Strangely enough, PrusaSlicer 2.5.0 works fine.  

The loading screen goes away too fast for me to see what text it says when it crashes.

There is no project file, as it crashes before the UI even starts.

When I run from the command line  (with loglevel6) I get this:

./PrusaSlicer --loglevel 6
[2023-02-18 21:18:58.537269] [0x00007ff858c3a680] [trace]   Initializing StaticPrintConfigs
[2023-02-18 21:18:58.627093] [0x00007ff858c3a680] [trace]   App updater default download path: "/Users/ameizeli/Downloads"
[2023-02-18 21:18:58.627258] [0x00007ff858c3a680] [debug]   full path: 1348679226012280357
[2023-02-18 21:18:58.627310] [0x00007ff858c3a680] [debug]   single instance: undefined. other params: ./PrusaSlicer;--loglevel;6
[2023-02-18 21:18:58.627338] [0x00007ff858c3a680] [debug]   full lock path: /Users/ameizeli/Library/Application Support/PrusaSlicer-alpha/cache/1348679226012280357.lock
[2023-02-18 21:18:58.627693] [0x00007ff858c3a680] [debug]   Creating lockfile.
[2023-02-18 21:18:58.627734] [0x00007ff858c3a680] [info]    Instance check: Another instance not found or single-instance not set.
[2023-02-18 21:18:58.798057] [0x00007ff858c3a680] [trace]   System language detected (user locales and such): en_US
[2023-02-18 21:18:58.802830] [0x00007ff858c3a680] [trace]   Best translation language detected (may be different from user locales): en
[2023-02-18 21:18:58.803070] [0x00007ff858c3a680] [trace]   Switching wxLocales to en
[2023-02-18 21:18:58.922986] [0x00007ff858c3a680] [info]    Checking if indices need to be installed from resources...
[2023-02-18 21:18:59.407384] [0x00007ff858c3a680] [info]    3dx mac handler starts
[2023-02-18 21:18:59.407561] [0x00007ff858c3a680] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file, not in dyld cache)
[2023-02-18 21:18:59.407573] [0x00007ff858c3a680] [info]    3dx drivers loaded: NO
[2023-02-18 21:18:59.411653] [0x00007ff858c3a680] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 108MB; Peak memory usage: 108MB
[2023-02-18 21:18:59.495936] [0x00007ff858c3a680] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2023-02-18 21:18:59.607722] [0x00007ff858c3a680] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2023-02-18 21:18:59.607746] [0x00007ff858c3a680] [debug]   GLCanvas3D: Use Retina OpenGL: true
[1]    21908 segmentation fault  ./PrusaSlicer --loglevel 6

----------

The report that is generated that OS X wants to send to apple is as follows:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [21908]
Path:                  /Users/USER/Desktop/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021 (PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021)
Code Type:             X86-64 (Native)
Parent Process:        zsh [2966]
Responsible:           iTerm2 [753]
User ID:               501

Date/Time:             2023-02-18 21:19:00.3018 -0600
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Bridge OS Version:     7.2 (20P3045)
Anonymous UUID:        481FC38E-1FBC-BD99-77B0-CB8D8FCCFDC3

Time Awake Since Boot: 52000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

[... 378 lines cut ...]
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 86.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 86.0

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-Alpha4

### Operating system

OS X Ventura 13.2.1

### Printer model

Prusa MK3S

kb9ysc, 2023-02-19:
A follow up note:  I wiped out my preferences in ~Library/Application Support/PrusaSlicer-alpha/  (deleted the whole directory in fact).   That seemed to do the trick, as Alpha 4 seems now to be starting up fine.   I zipped up the prefs folder if anyone wants me to upload it.  But for now I'm closing this as it doesnt seem likely it's an issue with the software itself.

closed 2023-02-19 by kb9ysc

lukasmatena, 2023-02-19:
If you can still reproduce it with the old configuration folder, we would be glad if you could share it. Thanks.

lukasmatena, 2023-02-19:
After reading #9774 - can you confirm that the crash happens even with our builds from github and not just your locally-compiled ones?

kb9ysc, 2023-02-19:
I've attached my preferences folder that was causing the issue.  and yes,
this crash occurs with the github builds and also my locally-compiled
ones.   I was able to reproduce the issue again by putting the attached
prefs back in place.  (tested with the github build.)

lukasmatena, 2023-02-19:
@kb9ysc There is no attachment. You have to upload it using the web interface, email attachments do not work. Thanks.

kb9ysc, 2023-02-19:
Hopefully this works:

[PrusaSlicer-alpha.zip](https://github.com/prusa3d/PrusaSlicer/files/10777129/PrusaSlicer-alpha.zip)

Thanks!

lukasmatena, 2023-02-20:
This seems to be the same issue as #9774. What is surprising to me is that you claim that wiping the config folder has helped. It should still crash when "show_hints" configuration option is true, and its default is true. Could you check whether your now running slicer has "Preferences->GUI->Show "Tip of the day" notification after start" enabled or not? Thanks.

kb9ysc, 2023-02-20:
Yes,  it is enabled.

I've uploaded a copy of my working prefs folder.  For this one, I made a copy of my 2.5.0 prefs folder and started up Alph4 with it--  but completely wiping the prefs also worked.

[PrusaSlicer-alpha-working.zip](https://github.com/prusa3d/PrusaSlicer/files/10783065/PrusaSlicer-alpha-working.zip)

kocikdav, 2023-02-20:
Please try this build. It is alpha4 build with just fix of this issue. 
MacOS:
https://slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-MacOS-universal-g6cb994706-202302201434.dmg
Linux:
https://slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-linux-x64-GTK3-g6cb9947-202302201436.AppImage

If it works and you are downloading from Printables.com, please try starting PrusaSlicer via the download button there. Unfortunately we do not poses a computer with this bug and I would like to test this works as well.

kb9ysc, 2023-02-20:
I tried to download the build, and it asked me for a username/password.  I tried to use my printables Username/password but that didn't work.

lukasmatena, 2023-02-20:
@kb9ysc Please use prusaslicer/slicer. Thanks.

kb9ysc, 2023-02-20:
Thanks.  It did work.  Here is what I did:

- backed up my  alpha settings (by renaming the folder).
- installed the alpha settings that didn't work above.
- tried to open standard Alpha4.  it crashed while starting up.  (testing that the issue was there.)
- I then ran the version with your fix. it started up, but the window was really strange (screenshot attached.).  turns out it was just shrunk,  I made the window larger,  and it looked normal. 
- Your version then prompted me for a configuration update  (I think Alpha1?  I forgot the version, sorry.)
- Closed out your version,  and opened up the alpha 4 version. It opened fine.

so something in the fix you did must have changed something in the settings such that the github Alpha4 no longer crashed at startup.  

[PrusaSlicer-alpha-post.zip](https://github.com/prusa3d/PrusaSlicer/files/10788390/PrusaSlicer-alpha-post.zip)
[image]

kb9ysc, 2023-02-20:
Forgot to add:  I also uploaded the settings file after the above steps were done.

regexident, 2023-10-28:
I'm pretty sure I'm seeing the same crash with both, v2.6.1 as well as v2.7.0-alpha1 (on macOS "Sonoma" 14.2 Beta (23C5030f)).

Somehow today PrusaSlicer stopped working as expected and is refusing to show an empty project window on launch, let alone open one when choosing "File > New Project".

When choosing "File > Open Project …" and selecting a project (any) it crashes the app.

1. Open "PrusaSlicer".
2. Choose "File > Open Project…" and select an existing project.
3. The app crashes.

```c
Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x110b63a95 ImFont::CalcTextSizeA(float, float, float, char const*, char const*, char const**) const + 69
1   PrusaSlicer                   	       0x110afd2d0 ImGui::CalcTextSize(char const*, char const*, bool, float) + 112
2   PrusaSlicer                   	       0x10f157174 Slic3r::GUI::NotificationManager::PopNotification::count_spaces() + 36
3   PrusaSlicer                   	       0x10f15bc8c Slic3r::GUI::NotificationManager::ProgressBarNotification::init() + 44
4   PrusaSlicer                   	       0x10f161d99 Slic3r::GUI::NotificationManager::SlicingProgressNotification::set_status_text(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 249
5   PrusaSlicer                   	       0x10f16aaff Slic3r::GUI::NotificationManager::set_slicing_progress_percentage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, float) + 351
6   PrusaSlicer                   	       0x10ee73898 Slic3r::GUI::Plater::priv::on_slicing_update(Slic3r::SlicingStatusEvent&) + 200
7   PrusaSlicer                   	       0x110ecc61e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
8   PrusaSlicer                   	       0x110ecc38b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
9   PrusaSlicer                   	       0x110ecc2a0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
10  PrusaSlicer                   	       0x110ecc11b wxEvtHandler::ProcessPendingEvents() + 235
11  PrusaSlicer                   	       0x110e25837 wxAppConsoleBase::ProcessPendingEvents() + 119
12  PrusaSlicer                   	       0x110ed62d2 wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 66
13  CoreFoundation                	    0x7ff8073ed796 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
14  CoreFoundation                	    0x7ff8073ed6ba __CFRunLoopDoObservers + 493
15  CoreFoundation                	    0x7ff8073ecc3c __CFRunLoopRun + 850
16  CoreFoundation                	    0x7ff8073ec322 CFRunLoopRunSpecific + 557
17  HIToolbox                     	    0x7ff811dbc9d9 RunCurrentEventLoopInMode + 292
18  HIToolbox                     	    0x7ff811dbc616 ReceiveNextEventCommon + 201
19  HIToolbox                     	    0x7ff811dbc531 _BlockUntilNextEventMatchingListInModeWithFilter + 66
20  AppKit                        	    0x7ff80a9f728d _DPSNextEvent + 880
21  AppKit                        	    0x7ff80b2eb7b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1304
22  AppKit                        	    0x7ff80a9e8803 -[NSApplication run] + 603
23  PrusaSlicer                   	       0x110deb68a wxGUIEventLoop::OSXDoRun() + 186
24  PrusaSlicer                   	       0x110ed691d wxCFEventLoop::DoRun() + 29
25  PrusaSlicer                   	       0x110e4d355 wxEventLoopBase::Run() + 85
26  PrusaSlicer                   	       0x110e250a4 wxAppConsoleBase::MainLoop() + 116
27  PrusaSlicer                   	       0x110dab3ca wxApp::OnRun() + 26
28  PrusaSlicer                   	       0x110e7e9f9 wxEntry(int&, wchar_t**) + 73
29  PrusaSlicer                   	       0x10ee038f8 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 248
30  PrusaSlicer                   	       0x10e0f2707 Slic3r::CLI::run(int, char**) + 21559
31  PrusaSlicer                   	       0x10e0f6f20 main + 352
32  dyld                          	    0x7ff806f89386 start + 1942
```

I tried deleting the configuration folder ("Help > Show Configuration Folder") in case it was corrupted, but even with a new and clean one the crash happens.

I was using the very same v2.7.0-alpha1 just fine the day before (and on the same macOS "Sonoma" 14.2 Beta (23C5030f)).