prusa3d/PrusaSlicer #9782: 2.6.0a4 locking up on export of G-Code (arcwelder script removed)

W1SPS, 2023-02-19:
### Description of the bug

Slicer locks up on export of G-code. Same problem as last time, but last time the problem was fixed by removing the AcrWelder script.  That has no effect this time.

### Project file & How to reproduce

[MK3s+2.6.0a4-KRM1&5.zip](https://github.com/prusa3d/PrusaSlicer/files/10776891/MK3s%2B2.6.0a4-KRM1.5.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 A 4

### Operating system

Windows 10 64bit

### Printer model

Prusa MK3s+

neophyl, 2023-02-19:
Just tried it and again no problems on my setup.  Alpha 4 and windows 10.  

Where are you saving the gcode to ?  I'm just saving it to the folder I extracted your zip into so there's no possible issues with network locations or somewhere else that it might have trouble accessing even temporarily.

W1SPS, 2023-02-19:
Uninstalled 2.5.0 restarted and used only 2.6.0a4 and it works even with arc welder.

closed 2026-09-29 by prusaslicer-bot[bot]