prusa3d/PrusaSlicer #9785: Feature request: Add support style on modifiers

alfredone78, 2023-02-19:
**Is your feature request related to a problem? Please describe.**
On some models it's better to choose a kind of supports in a specific part of the model and another kind of supports in another part of the model.
Using the modifiers it is not possible to choose the support style (Grid, Snug, Organic).

**Describe the solution you'd like**
I'd like to have the option "Support style" on modifiers options.

**Describe how it would work**
Just add "Support Style" on modifier options.

**Describe alternatives you've considered**
None.

**Additional context**
None.

pepperav, 2023-02-19:
Yeah cool idea! It should be just after "Layer and Perimeters" imho!

[image]

liftbag, 2023-02-19:
Nice idea.
Maybe it's not that simple due to the fact that organics are synchronized to layer height and that they don't support variable layers, but I hope a solution can be found, because for some objects it would be a very useful option.

it9gaj, 2023-02-19:
Nice idea

Ach-Mav, 2023-02-19:
Great suggestion.

zabbidou, 2023-12-06:
Commenting so hopefully it brings some more attention to this. I just encountered a model that desperately needs this feature. Until then, I sadly have to either painstakingly remove organic supports from a short flat base, or split the model into two more printable parts...

Tupson444, 2026-09-12:
still relevant

Jan-Soustruznik, 2026-10-20:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #16091.

closed 2026-10-20 by Jan-Soustruznik