prusa3d/PrusaSlicer #9787: Modifier Fuzzy skin perimeter issue

DVSVIDEO, 2023-02-19:
### Description of the bug

The modifier creates unnecessary walls when the fuzzy skin is turned on

### Project file & How to reproduce

[image]
This is when you have fuzzy skin (this is the error)
[image]
And this if other infill (this is good)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

win10

### Printer model

my own build

neophyl, 2023-02-19:
This is a duplicate of so many issues.  Please search both open and closed issues before opening new ones.   

I think all of them except this one https://github.com/prusa3d/PrusaSlicer/issues/6881 have been closed due to being duplicates.

DVSVIDEO, 2023-02-19:
I think this error has been reported since September 2021.
I have reported it again because I need a printout where this error is visible and prints a wrong object.

DVSVIDEO, 2023-02-19:
Since the fuzzy skin effect only makes sense on the outer surfaces of models, it should be taken out of the modifier settings on the inside of the object and put on a painting tool instead, just like support and seam painting.
(Hand painting, painting shapes.)

Snuff1eupagus, 2023-02-20:
@DVSVIDEO 
You can already do what you and so many others are asking for not paint them on but you can apply them without the extras walls. There are threads on the forum for this.

DVSVIDEO, 2023-02-21:
Which forum?

Snuff1eupagus, 2023-02-22:
The Prusa forum of course. 

Here's a link to one of the threads. About 8 posts down a user leaves a working example showing two different ways to achieve the result your looking for. Here's a hint, It's all about using volumes to control, as with any of the modifier meshes. I would look at the more complex example for enlightenment. Once it's set up you can move the modifier that applies the fuzzy skin anywhere you want on the models sides and get the effect. You can even apply different Fuzzy Skin settings to different areas if you setup your blocking Volumes and application Volumes correctly. Once you get a grasp of how it's working, you should be able to apply the concept when designing your future projects.  A brief explanation of the Object menu Hierarchy is provided as well.

https://forum.prusa3d.com/forum/prusaslicer/fuzzy-skin-modifier/

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena