prusa3d/PrusaSlicer #9790: Top and bottom layer issue

DVSVIDEO, 2023-02-19:
### Description of the bug

When the top and bottom layer is set to zero, infill is not rendering (except lightning infill).

### Project file & How to reproduce

[image]
The issue
[image]
This normal behavior (Super Slicer 2.5.59)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

Win10

### Printer model

Micromake C1 modified core xz printer

neophyl, 2023-02-19:
I just tried a cube in alpha 4 on windows 10 and with the closest I could get to your pictured settings
[image]
and I get infill.  

So could you please actually attach a project file as requested when opening an issue.  There must be some particular setting or combination of settings etc that is causing you the problem.  Even though you think its easy to reproduce, its not.  The request for attaching a project is not done for fun, its a vital part of trying to reproduce issues.

DVSVIDEO, 2023-02-19:
You are absolutely right:
[xyzCalibration_cube_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/10778005/xyzCalibration_cube_fixed.zip)

neophyl, 2023-02-19:
You have Only infill where needed ticked.  Print Settings>Infill>Reducing Print Time>Only Infill Where Needed.

Why it thinks its doesn't need infill I don't know but that's the setting that is stopping you getting the expected infill.

DVSVIDEO, 2023-02-19:
OOoops!
Thank you very much. I missed that little ticked box.
That was indeed the mistake.
Thanks again!

DVSVIDEO, 2023-02-20:
Solved!

closed 2023-02-20 by DVSVIDEO