prusa3d/PrusaSlicer #9793: Alpha4 won't open on mac os ventura

tyrscgm, 2023-02-19:
### Description of the bug

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [69519]
Path:                  /Volumes/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021 (PrusaSlicer-2.6.0-alpha4+MacOS-x64-202302180021)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-19 12:46:13.1370 -0800
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Anonymous UUID:        5DB06C4C-FF6F-45E1-B27D-19904A5F3FFB

Time Awake Since Boot: 54000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000014
Exception Codes:       0x0000000000000001, 0x0000000000000014

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [69519]

VM Region Info: 0x14 is not in any region.  Bytes before following region: 140737488273388
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffffec000-7ffffffed000 [    4K] r-x/r-x SM=SHM  

Kernel Triage:
VM - Returned success with no page
VM - Returned success with no page
VM - Returned success with no page
VM - Returned success with no page
VM - Returned success with no page

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x106928625 ImFont::CalcTextSizeA(float, float, float, char const*, char const*, char const**) const + 69
1   PrusaSlicer                   	       0x1068c1eb0 ImGui::CalcTextSize(char const*, char const*, bool, float) + 112
2   PrusaSlicer                   	       0x104fad464 Slic3r::GUI::NotificationManager::HintNotification::count_spaces() + 36
3   PrusaSlicer                   	       0x104faef3d Slic3r::GUI::NotificationManager::HintNotification::init() + 29
4   PrusaSlicer                   	       0x104fafe99 Slic3r::GUI::NotificationManager::HintNotification::retrieve_data(bool) + 425
5   PrusaSlicer                   	       0x104f7eddd Slic3r::GUI::NotificationManager::HintNotification::HintNotification(Slic3r::GUI::NotificationManager::NotificationData const&, Slic3r::GUI::NotificationManager::NotificationIDProvider&, wxEvtHandler*, bool) + 221
6   PrusaSlicer                   	       0x104f7bb56 Slic3r::GUI::NotificationManager::push_hint_notification(bool) + 310
7   PrusaSlicer                   	       0x104c61a60 Slic3r::GUI::GUI_App::post_init() + 1424
8   PrusaSlicer                   	       0x104c83273 wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_6>::operator()(wxEvtHandler*, wxEvent&) + 67
9   PrusaSlicer                   	       0x106c8ff9e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
10  PrusaSlicer                   	       0x106c8fd0b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
11  PrusaSlicer                   	       0x106c8fc20 wxEvtHandler::ProcessEvent(wxEvent&) + 96
12  PrusaSlicer                   	       0x106be8c0a wxAppConsoleBase::ProcessIdle() + 74
13  PrusaSlicer                   	       0x1069fce76 wxAppBase::ProcessIdle() + 22
14  PrusaSlicer                   	       0x106b6ecfa wxApp::ProcessIdle() + 26
15  PrusaSlicer                   	       0x106c99c6c wxCFEventLoop::OSXCommonModeObserverCallBack(__CFRunLoopObserver*, int, void*) + 92
16  CoreFoundation                	    0x7ff81708c3e1 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
17  CoreFoundation                	    0x7ff81708c309 __CFRunLoopDoObservers + 482
18  CoreFoundation                	    0x7ff81708b992 __CFRunLoopRun + 1177
19  CoreFoundation                	    0x7ff81708ae7f CFRunLoopRunSpecific + 560
20  HIToolbox                     	    0x7ff820f1d766 RunCurrentEventLoopInMode + 292
21  HIToolbox                     	    0x7ff820f1d576 ReceiveNextEventCommon + 679
22  HIToolbox                     	    0x7ff820f1d2b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
23  AppKit                        	    0x7ff81a1162f3 _DPSNextEvent + 909
[... 336 lines cut ...]
USB Device: USB20Bus
USB Device: hub_device
USB Device: FaceTime HD Camera (Built-in)
Thunderbolt Bus: MacBook Pro, Apple Inc., 23.4

### Project file & How to reproduce

starting the install 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha4+MacOS-universal-202302180021

### Operating system

mac os 13.2.1 (22D68)

### Printer model

ender3 klipper

lukasmatena, 2023-02-20:
Closing as a duplicate of #9774.

closed 2023-02-20 by lukasmatena

kocikdav, 2023-02-20:
Please try this build. It is alpha4 build with just fix of this issue. 
MacOS:
https://slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-MacOS-universal-g6cb994706-202302201434.dmg
Linux:
https://slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-linux-x64-GTK3-g6cb9947-202302201436.AppImage

If it works and you are downloading from Printables.com, please try starting PrusaSlicer via the download button there. Unfortunately we do not poses a computer with this bug and I would like to test this works as well.

tyrscgm, 2023-02-20:
Unauthorized
This server could not verify that you are authorized to access the document requested. Either you supplied the wrong credentials (e.g., bad password), or your browser doesn't understand how to supply the credentials required.

Apache/2.4.54 (Debian) Server at slicerbuilds.prusa3d.com Port 443

kocikdav, 2023-02-21:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-linux-x64-GTK3-g6cb9947-202302201436.AppImage
 https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.6.0-alpha4+1-dk-hints-public-MacOS-universal-g6cb994706-202302201434.dmg
or enter
user: prusaslicer
password: slicer
if dialog shows.