prusa3d/PrusaSlicer #9795: text embross input dialog doesn't allow pasting in text from clipboard

jvgogh, 2023-02-19:
### Description of the bug

when adding text to an object using the embross dialog, it doesn't allow pasting in text from clipboard

### Project file & How to reproduce

-select object, add part, text
-in embross dialog try to paste in text from clipboard
-no text is inserted into dialog

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 alpha 4

### Operating system

MacOS 12.6

### Printer model

Prusa Mini+

FidelCapo, 2023-02-20:
Thank you for the report. I am able to reproduce the problem on macOS.
Will try to find out why paste is not working in ImGUI dialogs and fix it.

FidelCapo, 2023-02-20:
SPE-1502

closed 2023-02-22 by FidelCapo