prusa3d/PrusaSlicer #9798: Connector holes filled after export to .STL

csterlin, 2023-02-20:
### Description of the bug

When using dowel connectors in split and attempting to export the resulting plate to a .STL file, the connector holes for the dowels are filled in.
These three images show the object before exporting to .STL file
[image]
[image]
[image]
These two images show the exported .STL file result, with holes missing, filled in.
[image]
[image]

### Project file & How to reproduce

This file shows the project right after slicing. I do not recall whether or not it was produced before or after exporting to .STL
[10mmCubeF50_Split_A4.zip](https://github.com/prusa3d/PrusaSlicer/files/10778299/10mmCubeF50_Split_A4.zip)
This file shows a separate project, the result of importing the previously exported file and the result of its slicing, showing the missing holes filled in.
[10mmCubeF50_Split_A4_Exported.zip](https://github.com/prusa3d/PrusaSlicer/files/10778300/10mmCubeF50_Split_A4_Exported.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0_Alpha4

### Operating system

Windows 11

### Printer model

Ender 3 (Original)

FidelCapo, 2023-02-20:
This happens because the holes for the connectors are created as negative volumes and are only subtracted from the model during slicing. When exporting the STL, the boolean operation with negative volumes is not performed.

There is already an issue for exporting STL and negative volumes, for example: https://github.com/prusa3d/PrusaSlicer/issues/7594

lukasmatena, 2023-04-02:
Subtraction of negative meshes during export was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena