prusa3d/PrusaSlicer #9800: Avoid Crossing Curled Overhangs Not Respecting Printer Bed Size

Gunner087, 2023-02-20:
### Description of the bug

The "Avoid crossing curled overhangs" option is generating travel moves outside of the dimensions of the printer.

### Project file & How to reproduce

Open project file, slice, and enable viewing travel moves in gcode preview. Toolpath exceeds the dimensions of the build plate at the upper right. Turn off "avoid closing curled overhangs" and the issue is resolved. Screenshot and project below.
[image]
[Avoid crossing curled overhangs bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10778738/Avoid.crossing.curled.overhangs.bug.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

macOS 13.0.1

### Printer model

Voron0 v0.1

Godrak, 2023-02-20:
Hi, thank you for the report. This should not certainly happen, we will look into it.

kubispe1, 2023-02-24:
Should be fixed and tested. I am closing this issue. Thanks