prusa3d/PrusaSlicer #9801: Alpha4 organic support is too exaggerated compared to Alpha3 (that was working fine)

bsas, 2023-02-20:
### Description of the bug

I am testing the new "organic support" on Alpha3 for a while with an amazing success rate for complex organic STL files. It was working great and using very few filament.

Today I updated to the Alpha4 and with all settings the same and something changed. The organic supports getting generated now are far more "everywhere", using almost 2x more filament. Didn't print to test yet, maybe it will also work, but I don't understand what changed from Alpha3 to Alpha4 that completely made the behavior go crazy :(

### Project file & How to reproduce

Example file: Mug (https://www.thingiverse.com/thing:3345027/files)
Just set to organic support (default values) and slice.

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6 Alpha3 to Alpha4 update

### Operating system

Windows 10 and MacOS

### Printer model

Voron 2.4 and Neptune 3

rcmaniac25, 2023-02-20:
Noticed the same thing. Tried to re-create the Benchy demo...
Alpha 3 - success
Alpha 4 - entombed benchy

There were certainly parts of prints in Alpha 3 that weren't fully supported, but Alpha 4 now over supports with the same settings. I've seen it with some other models I've sliced and even need to make an update to my #9621 ticket because the comparison is quite different now.

There's still a chance that it's just that organic support defaults need updating, but for right now it's excessive for many models I want to use organic supports on. I'll probably use Alpha 3 for those as of now.

OopAck, 2023-02-20:
I agree that a4 organic supports are too exaggerated compared to a3.
I have gone back to a3 build.

Please address this in the next build.

JohnOCFII, 2023-02-21:
I too noticed the large difference between alpha-3 and alpha-4.  At first I assumed there would be obvious settings changes, but the settings appeared identical.  Here's one example:
[image]
[image]

MrBlackmidi, 2023-02-23:
having this issue too, hope it gets fixed

bsas, 2023-02-23:
I am back to Alpha3 until this is fixed. I really wish it is fixed before the final release.
My impression is that on Alpha3 the code realizes what is already supported on each branch of the tree support to avoid create unnecessary extra branches, but Alpha4 doesn't. So, it keeps trying to support stuff that is already supported. That is my guess based on try and error...

kb-, 2023-02-23:
The way it's done on the top picture looks right. Overhangs are supported.
The bottom one doesn't look right. The few spotty supports seem useless. The overhang looks the same width all around the part, so it doesn't make sense to only have support at these locations. 
[image]

Having less dense support should be handled by "branch density", but when I compare the default 30% VS 5% I don't see a change.

race85com, 2023-02-24:
I have the same issue, Alpha4 generating way too many supports on the exact file & config compared to Alpha3

rcmaniac25, 2023-02-27:
@ThomasRahm I agree Alpha 3 is under supported and thank you for basically kickstarting all the major slicers implementing tree supports with your fork. Here's my view on it...

I think the challenge is the point of organic supports (or as named in your original fork: improved tree supports), which is
> "Tree supports" - support structures designed to save material and printing time as much as possible. Branches of supports are generated where needed and limit contact areas to a necessary minimum.
(taken from the 2.6 alpha 2 release notes)

With the included image:
[image]

And a vast majority of people are going to use defaults.

This is what we were proposed with turning on organic supports:
[image]
Prints in 2h 9m (14.1g)

This is what we got (under alpha 4, best attempt to get the same angle as Prusa's since they didn't release the 3mf):
[image]
Prints in 2h 48m (17.93g)

Only thing turned on was organic supports with build plate only. With snug supports, this is 2h 37m (17.24g), and original grid is 2h 54m (19.43g).

So it doesn't save time and material _with existing defaults_. If I change the overhang threshold to 20 degrees, I get something much closer to the original:
[image]
Prints in 2h 9m (14.3g)

The bottom isn't really supported and I expect this to fall over before it even gets to the supports, but visually it's similar. A better value was 25 degrees, which was 2h 10m and 14.3g, which supported the bottom and back letters better.

Since tradition is "over 45 degrees is risky" and the default overhang threshold for PrusaSlicer is 50 degrees produces a result that does not meat the proposed definition of tree supports (it uses more material and takes longer to print), I think the default should be changed (at least for tree supports). The introduction of "avoid crossing curled overhangs" and the intelligent automatic paint on supports makes me think that it should be possible to use existing defaults but be able to determine "does this point actually need supports?" as opposed to simple overhang angle (I have not seen the code, I don't know what other factors are taken into account right now. I just see overhang angle having an almost inverse effect on the supports)

My end goal vision is to turn on organic supports and be done, which is something I believe a vast majority of people will do. Get the benefits of reduced print times and reduced filament usage while limiting contact area. Alpha 3 under supported by default and I had at least one test print fail because of that. Alpha 4 under defaults overly supports.

Does that make sense?

Shameless plug: my other ticket #9621 uses support enforcers, a setting that does not utilize overhang threshold and somehow manages to add more supports then even this, and which automatic painting also doesn't utilize an overhang threshold from the tool. And since the other support modes add _additional_ support, along with support enforcers. It makes support enforcers basically unusable with organic supports. So if the solution here is "just adjust your overhang threshold", then what happens what that isn't available? Is it suffering from the same underlying issue as this ticket?

DerGenaue, 2023-02-27:
The Bug fixed in alpha4 is this one btw: https://github.com/prusa3d/PrusaSlicer/issues/9459 / https://github.com/prusa3d/PrusaSlicer/issues/9556
The Alpha 3 did not generate enough supports due to a bug in the overhang detection.
Alpha 4 now generates supports everywhere where the settings ask it to.

> My impression is that on Alpha3 the code realizes what is already supported on each branch of the tree support to avoid create unnecessary extra branches, but Alpha4 doesn't. So, it keeps trying to support stuff that is already supported. That is my guess based on try and error...

That was not the case, it was just a stupid error in the calculations.  
Smart support placement is available though: **use the "Automatic FDM support painter"** that also was introduced in the new Alpha.
It does pretty much exactly what you described.

The `Branch Density` setting should allow you to control the amount of branches generated.

bsas, 2023-02-27:
I am just curious if there is any setting that can make it go back to similar levels of filament and time consumption as Alpha3, even with the "risk" of being under supported...

ThomasRahm, 2023-02-28:
@rcmaniac25 "I just see overhang angle having an almost inverse effect on the supports" A horizontal plane is not a 90° overhang in PrusaSlicer, but 0° and a vertical plane is a 0° overhang. I just say this as `overhang angle` is the terminology Cura uses, where it is the other way round. So it has an inverse effect on support.  

I disagree that it over-supports. It supports the same areas that non organic support also supports. 
By that logic non organic support also over-supports. 
That may be a valid opinion, but I am not knowledgeable enough to have a factual opinion of what the default should be, as i only have experience with one fdm printer.

@DerGenaue
The reason that the branch density does not work here is that every overhang will be supported, and as the model is not flat, there is not one overhang area per layer, but tens/hundreds. The tip placement is calculated for each layer independent from the other layers.
Branch density only influences how many tips are placed below an overhang, but there is a lower limit (if i remember correctly it is 3) to ensure every overhang has at least some support, so if the overhang areas are small enough, the branch density will not change the result.

@bsas One of the bugs that were fixed, caused the placing of the center of a tip further away from the model based on the tip diameter to not work correctly, effectively increasing the limit of overhang that could be supported. 
So reducing the `overhang threshold` should do practically the same, because the distance the tips have to be from the model and the overhang threshold are logically related (If the tip has to be a certain distance from the model in the xy plane, and a certain distance in the z plane, only a overhangs up from a certain angle can be supported, because _tip_, _model on the same layer as the tip_ and _model supported by the tip_ is a triangle)

rcmaniac25, 2023-02-28:
@ThomasRahm I thought about the what I wrote and alluded to it at the end, but my point is "defaults matter as many won't be changing them. and the defaults, such as overhang threshold/angle, don't give the desired outcome even if it's the correct outcome" and think that it should be adjusted for organic supports. And yes, non-organic supports also over-support. That's why there was a lot of demand for organic supports, because it was (a lot) less supports.

closed 2026-09-29 by prusaslicer-bot[bot]