prusa3d/PrusaSlicer #9805: Text embossing tool - Right to left (RTL) languages

masu100, 2023-02-20:
### Description of the bug

Text embossing tool is writing the reverse order a right to left languages,
so for print it correctly, I need to type it in reverse order.

In the screenshot, can be seen the correct direction in the notepad:
[image]

### Project file & How to reproduce

1. add .stl file
2. add Text embossing 
3. type a text in a right to left language 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

Windows 10 Pro (English)

### Printer model

Prusa MINI+

Jony01, 2023-02-20:
We can't detect RtL font. What more our GUI framework did not support writing RtL(it has right reason why not to support)
For reverse order of character in text you could use some online tool e.g. [[link](http://transdict.com/tools/cs-torevert.html)]

I am open for pull request on this feature request.

atg90, 2023-04-19:
still this is not helpful, since issue is not only reversing letters, but not connecting them as they should in languages like Arabic.
[image]

Jony01, 2023-04-20:
I am not familiar with Arabic writting. Could you provide better example with TTF font file you are writing, unicode symbols you type and how it should looks like(image)?

atg90, 2023-04-23:
@Jony01 thank you Filip for trying to help or figure out how can we fix this..
i will be really glad to collaborate closely to help the community resolve this issue for affected languages.

here i am attaching photo showing all the details and comparison. i can provide further clarification if requested.

note that arabic letters can change their form and slightly depending on whether they are connected at beginning/ middle or end of a word.

Windows fonts when typing takes care of doing this by itself.. i dont know if its the TTF file or some other predetermined settings in windows. but yes. it seems every letter will be having 4 forms in TTF file:
1. standalone 2. beginning 3. middle or 4. end of a word.
[PrusaSlicer Arabic Text Emboss issue.docx](https://github.com/prusa3d/PrusaSlicer/files/11303357/PrusaSlicer.Arabic.Text.Emboss.issue.docx)

[image]

ymca11, 2023-05-14:
> We can't detect RtL font. What more our GUI framework did not support writing RtL(it has right reason why not to support) For reverse order of character in text you could use some online tool e.g. [...]

I also give 👍 for this feature.

UrielGuy, 2023-11-25:
+1 here

RedTrainerF, 2026-09-02:
Alright i guess i have to refresh this.... the problem still lives on until today(version 2.9.6).
writing stuff in english and other compatible languages is a breeze and fun, but in arabic? it makes the 5 seconds task into 3 minutes task.

in prusa... letters arabic letters are both split and also reversed, i know this is a tough language to handle but... i really hope this is fixed in the future, the sooner the more thankful i will be.

[image]

encip, 2026-09-02:
Issue still exists in PrusaSlicer 3.0.0 alpha.  Text should appear as **גן** and **بيت** but instead it looks like this, with the letters backwards in both languages, and in the case of Arabic, in their separate/independent forms instead of connected)

[image]

[image]