prusa3d/PrusaSlicer #9807: Bridge Infill missing with Generate Support Material enabled and Don't Support Bridges enabled.

Quik, 2023-02-20:
### Description of the bug

When Generate Support Material is enabled and Don't Support Bridges is enabled or disabled and the model is sliced bridge infill is not created:

[image]

When Generate Support Material is disabled:

[image]

### Project file & How to reproduce

[temp_tower_custom_230.to.255.zip](https://github.com/prusa3d/PrusaSlicer/files/10785228/temp_tower_custom_230.to.255.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1-alpha4-win64

### Operating system

Windows 10

### Printer model

creality ender-2 s1 pro

bubnikv, 2023-03-03:
You have set the support gap to zero, meaning a soluble support interface material. In such case PrusaSlicer expects the bottom surfaces to be supported by soluble interfaces and it does not apply the bridging logic.

Did you select the soluble supports intentionally?

Quik, 2023-03-30:
Thank you for the feedback.  Yes, the support gap parameter was set to zero intentionally because I have been using organic supports and this was the recommendation for the alpha release as far as I understood.

closed 2026-09-29 by prusaslicer-bot[bot]