prusa3d/PrusaSlicer #9809: PrusaSlicer 2.6.0 alpha4 crashed when manipulating objects

the-ress, 2023-02-20:
### Description of the bug

Crash dump: [prusa-slicer.exe.11304.zip](https://github.com/prusa3d/PrusaSlicer/files/10785542/prusa-slicer.exe.11304.zip)

### Project file & How to reproduce

1. I pressed Ctrl+O and opened this 3mf file: [tank.zip](https://github.com/prusa3d/PrusaSlicer/files/10785495/tank.zip)
2. I've hidden all objects
3. I pressed 'A' to move them out of the way
4. I duplicated `tank main body (no tires) (1).stl` (Ctrl+C, Ctrl+V)
5. I checked the eye icon next to the copy
6. I probably pressed 'A' to move the copy to the build plate
7. I might have pressed 'Split to parts', or PS crashed before I could do that

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

Windows 11 (22621.1265)

### Printer model

MK3S+

the-ress, 2023-02-20:
I reproduced the issue: [prusa-slicer.exe.27196.zip](https://github.com/prusa3d/PrusaSlicer/files/10785566/prusa-slicer.exe.27196.zip)

This time I did the same steps, including clicking 'Split to parts', then I clicked one or more parts.

I'll try to capture the full dump.

the-ress, 2023-02-20:
Here's the full memory dump: https://tomcova-my.sharepoint.com/:u:/g/personal/tereza_tomcova_onmicrosoft_com/Ef1J2vJUstdGpae9ZzWsMtkBaH7E_xztNKFR6UtRn3oG2w?e=7wwXTL
(There should be a download button in top left corner.)

It only crashed after I kept clicking the parts randomly for a few seconds.

kubispe1, 2023-02-20:
Thanks, we can reproduce it. Thanks for the whole description. The TriangleSelector function probably fails.

FidelCapo, 2023-02-28:
Thank you for the report. This issue should be fixed in the next release.
Closing.

closed 2023-02-28 by FidelCapo