prusa3d/PrusaSlicer #9811: fullcontrol.xyz gcode not rendered properly

MrBrain295, 2023-02-20:
### Description of the bug

When I open gcode from fullcontrol.xyz it does not render correctly. A correct rendering of what it prints as.
[image]

### Project file & How to reproduce

1. Go to fullcontrol.xyz
2. Download the pin support challenge.
3. Open it in Pursa slicer gcode viewer.
4. See it incorrectly rendered.
[image]
NOTE: The print did work correctly with that exact gcode.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+linux-x64-GTK3

### Operating system

Ubuntu 22.10

### Printer model

Ender 3 V2 Neo

kubispe1, 2023-02-20:
Hi, please check this issue and eventually close it. I think it is a duplicate and the bug has been fixed. #9099
For new bug reporting. Please use 2.6.0-alfa4 version. 
Thanks

MrBrain295, 2023-02-20:
Sorry, I only looked in the open issues didn’t realize it was already fixed.

closed 2023-02-20 by MrBrain295