prusa3d/PrusaSlicer #9812: Random Travel Moves

NJFoggio, 2023-02-20:
### Description of the bug

Multiple times I have had random travel moves during prints once crashing during a multi-model print (one at a time). I've attached this most recent G-Code file that is quick and has an example of this issue. 

### Project file & How to reproduce

[1,75-Clip Bug Files.zip](https://github.com/prusa3d/PrusaSlicer/files/10787655/1.75-Clip.Bug.Files.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11

### Printer model

Prusa mini plus

NJFoggio, 2023-02-21:
Another example. If this is a setting I would love to know which one.
[a080071-bl_0.15mm_PLA_MINI_5h14m.zip](https://github.com/prusa3d/PrusaSlicer/files/10789293/a080071-bl_0.15mm_PLA_MINI_5h14m.zip)

NJFoggio, 2023-02-21:
@lukasmatena. It has occurred with both the original USB and my new lower profile USB. It's properly formatted and has worked flawlessly sometimes. I will be preforming some more troubleshooting after I am off from work today to see if it is due to using the Ethernet to upload files. Hopefully this will narrow the problem down to the uploading or to the USB or printer itself. Also is there a guide on resetting prusa slicer back to factory defaults just so I can be sure nothing is wrong with the slicer.

lukasmatena, 2023-02-21:
@NJFoggio If you want to try "resetting" PrusaSlicer configuration, then go to `Help->Show configuration folder` (an explorer window will open), close PrusaSlicer and erase the whole PrusaSlicer (or PrusaSlicer-alpha) directory which the explorer shows. It is just the configuration directory, not where the application itself lives. When you next start PrusaSlicer, it will recreate the configuration folder from defaults (you will get a wizard on the first start). Note that this action will remove all your user profiles etc., so you should backup the folder before (unless you're using everything stock).

I don't think the problem is caused by anything in the configuration, though. In theory there might be an issue in how we export the file to the drive or how we eject it, so you may want to save the file to your local disk first and eject the drive through the system to rule these two out.

Please, let us know how it went.

NJFoggio, 2023-02-22:
After Re-Partitioning the USB and Resetting both PrusaSlicer and my Prusa Mini + the upload of G-Code works without issue.

lukasmatena, 2023-02-22:
Thanks for the update, I'm glad it's sorted out. Closing the issue.

closed 2023-02-22 by lukasmatena