prusa3d/PrusaSlicer #9813: When drawn supports are used, the option "support from table only" is ignored

yarus23, 2023-02-20:
### Description of the bug

When drawn supports are used, the option "support from table only" is ignored

### Project file & How to reproduce

[smiling_toothbrush_holder_simple_partition.zip](https://github.com/prusa3d/PrusaSlicer/files/10787787/smiling_toothbrush_holder_simple_partition.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4+win64

### Operating system

Windows 11

### Printer model

Wanhao D12/300

bubnikv, 2023-03-03:
That is on purpose. Actually people reported when it worked as you want it.

jmd73, 2023-03-16:
@yarus23 There are errors in you object. After repairing it with netfabb and trying to recreate the painting, it seems ok.

I think @bubnikv did not see the supports generated inside the object.

[smiling_toothbrush_holder_simple_partition_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/10989695/smiling_toothbrush_holder_simple_partition_repaired.zip)

closed 2026-09-29 by prusaslicer-bot[bot]