prusa3d/PrusaSlicer #9814: crash on slicing this file at generating support material.

usmarine0622, 2023-02-20:
### Description of the bug

It crashes when slicing when it gets to the generating support stage when using organic supports. When using grid supports it doesn't crash.

### Project file & How to reproduce

[Alien - Full Figure AMENDED.zip](https://github.com/prusa3d/PrusaSlicer/files/10787891/Alien.-.Full.Figure.AMENDED.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha4

### Operating system

Windows 11 Pro

### Printer model

Original Prusa i3 MK3S+

kubispe1, 2023-02-21:
Sorry, but after some basic tests, we are not able to reproduce it on WIN 10, WIN 11, and Linux Asan. 
Please could you provide us more info - detailed steps you did, eventually your pc configuration?

usmarine0622, 2023-02-21:
I created a video of what it's doing when I try to slice the file using organic supports. I opened the project file I uploaded in the initial post and hit slice. It looks like it's working but the program just close without any error messages. My system specs are Intel I9 13900K, 64GB DDR5 ram, RTX 4090 using the latest drivers.

[2023-02-21 09-16-37.zip](https://github.com/prusa3d/PrusaSlicer/files/10794947/2023-02-21.09-16-37.zip)

usmarine0622, 2023-02-21:
> I created a video of what it's doing when I try to slice the file using organic supports. I opened the project file I uploaded in the initial post and hit slice. It looks like it's working but the [...]

I created a video of what it's doing when I try to slice the file using organic supports. I opened the project file I uploaded in the initial post and hit slice. It looks like it's working but the program just close without any error messages. My system specs are Intel I9 13900K, 64GB DDR5 ram, RTX 4090 using the latest drivers.

[2023-02-21 09-16-37.zip](https://github.com/prusa3d/PrusaSlicer/files/10794958/2023-02-21.09-16-37.zip)

usmarine0622, 2023-02-21:
> Sorry, but after some basic tests, we are not able to reproduce it on WIN 10, WIN 11, and Linux Asan. Please could you provide us more info - detailed steps you did, eventually your pc configuration?

I created a video of what it's doing when I try to slice the file using organic supports. I opened the project file I uploaded in the initial post and hit slice. It looks like it's working but the program just close without any error messages. My system specs are Intel I9 13900K, 64GB DDR5 ram, RTX 4090 using the latest drivers.

[2023-02-21 09-16-37.zip](https://github.com/prusa3d/PrusaSlicer/files/10794967/2023-02-21.09-16-37.zip)

sosso, 2023-02-23:
Similar issue for me using Organic supports on https://www.printables.com/model/165092-frosthaven-shambling-skeleton at a 0.05 layer height. Grid supports work fine, organic supports choke a 5800x3D (100% CPU usage) processor and the app hangs at 70% in Generating Support Material.

bubnikv, 2023-03-03:
@sosso You may be low on RAM for such a model and 0.05 layer height.

kubispe1, 2023-08-29:
@usmarine0622, @sosso please can you check the issue with 2.6.1-rc1.
Thanks a lot for your cooperation.

lukasmatena, 2023-11-21:
No response, no way to reproduce. Closing.

closed 2023-11-21 by lukasmatena