prusa3d/PrusaSlicer #9815: Arrange object - add spacing option and rotation option

Eldenroot, 2023-02-20:
**Is your feature request related to a problem? Please describe.**
I need to get these two settings when I print multiple instances on one plate.

**Describe the solution you'd like**
[image]

**Describe how it would work**
Add a new group of settings - rotate objects + spacing between objects.

**Describe alternatives you've considered**
None.

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2023-02-21:
Already exists in PS , right click the arrange tool.

Eldenroot, 2023-02-22:
OK, srry. Closing :)

closed 2023-02-22 by Eldenroot