prusa3d/PrusaSlicer #9816: neither raft or brim  generated in case if organic support is used

yardu1, 2023-02-21:
### Description of the bug

it's impossible to use simultaneously organic support, brim and raft or organic support and raft. 

### Project file & How to reproduce

use model that requires support.
use organic support
choose Brim check mark
add 3 layers of raft

gcode will omit both raft and brim.

if raft not configured - brim will be generated as excpected. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alfa4+Macos-x64

### Operating system

mac osx ventura 13.1

### Printer model

prusa mks 3+

foreachthing, 2023-02-21:
See: https://github.com/prusa3d/PrusaSlicer/issues/9526
and: https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2

> The Organic supports currently pose the following limitations:
> 
> For large models with a lot of supports, 16GB RAM minimum is recommended.
> Top / bottom vertical separation distances are rounded to multiples of layer height.
> Variable layer height is not supported.
> Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.
> Rafts with Organic supports are buggy.

kubispe1, 2023-03-01:
Hi, thanks for the issue. In the next release, there should work raft OR brim properly in combination with organic supports. If you find some complications after release 2.6.0-alfa5 will be release let us know. Also, see our changelog that will come with.
Closing the issue.

closed 2023-03-01 by kubispe1