prusa3d/PrusaSlicer #9819: No organic support when standard supports exist

F-CJPJ, 2023-02-21:
### Description of the bug

Hello

I don't know if it's a bug or  feature.
I have sliced a shaoe where I need standard supports on the plate and would like to have organic support for 2 elements of my shape which needs support (not from the plate).
I attach the screen shot where standard supports is shown with the 2 elements identified as needing support but not supported.
The check box "support only from plate" is checked.
[image]

### Project file & How to reproduce

[Meli-Manchon_bas_v5.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10793327/Meli-Manchon_bas_v5.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6

### Operating system

macOS13.1  arm

### Printer model

wanhao D12

F-CJPJ, 2023-02-21:
version 2.6.0 alpha4

lukasmatena, 2023-02-21:
This is basically a duplicate of #9785. Closing.

closed 2023-02-21 by lukasmatena

lukasmatena, 2023-02-21:
We will leave #9785 opened, but implementing this is not currently on the roadmap. It is very difficult to do.