prusa3d/PrusaSlicer #9822: Reload from disk unavailable for 3mf files in 2.6.0-alpha4

dimitrisch, 2023-02-21:
### Description of the bug

The "Reload from disk" model preview and object list right click menu option is disabled for 3mf models in latest 2.6.0-alpha4. 
It is enabled for all other file types (OBJ, STEP, STL, AMF) and the Edit > Reload from Disk menu option ignores 3mf files.
In 2.5.0 I can reload my 3mf models correctly.

### Project file & How to reproduce

[3mf-test.zip](https://github.com/prusa3d/PrusaSlicer/files/10798099/3mf-test.zip)
The attached file contains a sample project and two sample 3mf models.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha4+win64

### Operating system

Windows 11 Pro 22621.1265 

### Printer model

Voron 0 in this project file but irrelevant

XoBoR, 2023-02-28:
I can confirm this. Really annoying.

kubispe1, 2023-03-03:
Hi, thanks for the report. For the upcoming release 2.6.0-alfa5 issue should be fixed. I am closing it for now. If you will have some problem let us know. 
Thanks

closed 2023-03-03 by kubispe1