prusa3d/PrusaSlicer #9823: G-Code Viewer 2.6.0 Alpha4 crashes with (modified) Simplify3D Version 4.1.2 gcode

dan-and, 2023-02-21:
### Description of the bug

Open gcode file from https://teachingtechyt.github.io/calibration.html#accel in the PrusaSlicer G-Code Viewer. 
Gcode loads just fine. 

When switching the View to "tool" , then 2.6.0-alpha4 crashes 

Same gcode runs fine with the gcode viewer in prusaslicer 2.5.0 

### Project file & How to reproduce

1. Start PrusaSlicer G-Code Viewer
2. Load accelleration.gcode file 
3. Switch view to "tool" 

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4

### Operating system

Windows 11

### Printer model

Klipper-based TT Saphire PRO SP3

dan-and, 2023-02-21:
[acceleration.zip](https://github.com/prusa3d/PrusaSlicer/files/10798668/acceleration.zip)

dan-and, 2023-02-21:
[image]

FidelCapo, 2023-02-27:
Thank you for the report.
This issue should be fixed and the fix will be included in the next release.
Closing.

closed 2023-02-27 by FidelCapo