prusa3d/PrusaSlicer #9826: Nozzle hitting supports (not avoiding them)

swheettaos, 2023-02-22:
### Description of the bug

Using organic supports.  It prints supports, then prints part, then goes to print the next layer, and if there are supports in the way it will bang them.  As the print goes on, after repeated hits, the support gets so out of whack that it ends up breaking the supports off and ruins the print.  I have "avoid crossing perimeters" enabled and I have also tried "avoid curling", with the same result.  I don't want to do a z-lift on every retraction.  I have also tried slowing the print speed on supports down a little thinking it was the quality of the supports causing them to be slightly higher.  No luck.  I had the problem a little bit when using old supports in PrusaSlicer 2.5 (especially if the print was a tall one) but it didn't break them off.  I think standard supports are just sturdier. I don't have a problem with it hitting the part (unless I have a bit of a curl) only the supports.

### Project file & How to reproduce

[Shell belt caddy with supports.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10800876/Shell.belt.caddy.with.supports.3mf.zip)

It hits the supports on the side for the lower part of the print where it gets in the nozzel path but doesn't break them (thicker still?). When it gets toward the end of the print the supports at the back (near the top of the clip) start to get in the path of the nozzle. They are then the ones that normally break after repeated hits.  In the front part of the print towards the middle height, I get rubbing but no braking. This last print, also broke the supports at the top of the print in the front as they got in the path of the nozzle.  

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha4+MacOS-x64

### Operating system

MacOS 13.2.1 (22D68)

### Printer model

Neptune 2 with Swiss Micro hotend, direct drive, Marlin 2.1.2, This also happens on my Ender 3 v2 also running Marlin 2.1.2 (it always has happened)

foreachthing, 2023-02-22:
You should be using z-hop/z-lift.

DVSVIDEO, 2023-02-22:
> It hits the supports on the side for the lower part of the print where it gets in the nozzel path but doesn't break them (thicker still?). When it gets toward the end of the print the supports at [...]

This problem is probably printer dependent. I had the same problem with my Cura Tree support when the cooling was not sufficient.
Then 100% cooling (high performance fan, stealthburner toolhead) solved this problem.
Insufficient cooling can cause the edge of the support to curve upwards, which then gets in the way of the nozzle. Notice when this happens.
This is not the fault of the slicer...

swheettaos, 2023-02-22:
ok.. if I have to use z-lift .. what is a good setting?  something small like .1 or .2?  shouldn't need much right?

foreachthing, 2023-02-22:
I've used 0.2 mm and that was sufficient so far.

swheettaos, 2023-02-22:
Ok, that seems to be working.  Is there a way to only have a z-lift when using supports?  I know I can create a new filament/printer profile with it enabled and switch between them when using supports, but I'd like to make the same profile work for all my prints and only have z-lift kick-in when printing with supports.  The z-lift on my system is only necessary when printing supports and it slows the print down. in the project I uploaded, it went from 7h30m to 8h0m with a .2 z-lift.  why slow it down if I don't need to?

foreachthing, 2023-02-22:
You could post process your gcode to check for supports and then leave it in. Otherwise remove all z-lifts.

swheettaos, 2023-02-22:
ok. I'll look into it.

closed 2023-02-22 by swheettaos