prusa3d/PrusaSlicer #9827: Connection between sphere and cylinder sliced wrongly

sbellon, 2023-02-22:
### Description of the bug

When a sphere touches a cylinder as shown in the screenshots below there is a groove (I hope that describes it best?) between the cylinder and the sphere.

The STL file does not exhibit that.

I tried slicing from ULTRADETAIL (0.05mm) to DRAFT (0.30mm), but it's always present. Also it's present in latest release 2.5.0 as well as the 2.6.0-alpha releases.

To me it looks like the algorithm follows more the shape than the cylinder and detects the intersection "too late" (or "too early" depending on how you look at it)?

### Project file & How to reproduce

[Triggerding.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10801066/Triggerding.3mf.zip)
[image]
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+linux-x64-GTK2, 2.6.0-alpha3+linux-x64-GTK2, 2.6.0-alpha4+linux-x64-GTK2

### Operating system

Debian bookworm GNU/Linux

### Printer model

Original Prusa i3 MK3S

lukasmatena, 2023-02-22:
Your model contains some weird intersecting geometry around there. It is buried inside the "conical" part and confuses the slicing algorithm which does not know which of the boundaries to follow. This is what I mean (I split the object into parts and moved the extra part slightly):
[image]

I am not sure why it is there, but it is enough to split your object to parts (in the top menu), remove the extra part and then slice.

sbellon, 2023-02-22:
Ouch, ok, so then I should rather file this as a bug against OpenSCAD.

Thanks for the lesson learnt: In the future I will use "split to ..." first in order to check whether something strange is going on.

:+1:

closed 2023-03-06 by Godrak