prusa3d/PrusaSlicer #9828: Add custom GCode in between layers just like you can add before a layer

Bumblebiber, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
I wanted to print that one layer of the left object slower, as its the first layer after an support interface layer. But the right object should print at normal speed.

**Describe the solution you'd like**
Add the +-Button on the lower slider

**Describe how it would work**
Same way as it does on the "right slider"

**Describe alternatives you've considered**
You can do this manually searching for the line on the left, then open the GCode file in VS Code and add the desired GCode.

[image]

lukasmatena, 2023-02-22:
This is certainly a duplicate of something, but I am unable to find the issue right now.

Regarding your use case. Is there a reason you don't just use the [Height range modifier](https://help.prusa3d.com/article/modifiers_1767#height-range-modifier) for that? It seems like a lot easier way of doing the same thing.
[image]

lukasmatena, 2023-02-22:
This is a duplicate of #5195

Bumblebiber, 2023-02-22:
> This is a duplicate of #5195

Hm... Almost 2 and a half years old but it didn't happen :( This is such a cool feature and would be super easy to implement.

lukasmatena, 2023-02-22:
Closing as a duplicate of #5195.

closed 2023-02-22 by lukasmatena