prusa3d/PrusaSlicer #9829: Add tags to different features in GCode

Bumblebiber, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
I can't set custom fan speeds e.g. for outer walls.

**Describe the solution you'd like**
If there was just a comment like "; outer wall", "; inner wall", "; infill" before the related feature, you could then use the GCode substitution function to add custom GCode. For example:

Search: "; outer wall" Replace with: "M106 S255"
Search "; infill" Replace with "M106 S0"

**Describe how it would work**
This way, you could only use the fan on outer walls and for the infill it would stay off, providing better layer adhesion for ABS and similar filaments. Just one example, you can also play with different settings than fan speed.

**Describe alternatives you've considered**
Use SuperSlicer. This one has the option for Wall Fan Speed.

lukasmatena, 2023-02-22:
Are you aware of `Print Settings->Output options->Verbose G-Code` option?

Bumblebiber, 2023-02-22:
> Are you aware of `Print Settings->Output options->Verbose G-Code` option?

I wasn't. Thank you, that's awesome :)

lukasmatena, 2023-02-22:
Ok, I guess that means the issue can be closed.

closed 2023-02-22 by lukasmatena