prusa3d/PrusaSlicer #9830: When using one "Top Solid Layer" fan goes to zero

onolox, 2023-02-22:
### Description of the bug

Filament: PLA

Hello, I think that it's a bug, when I set to Top Solid Layer to 1 it makes the layer to be printed without fan at all.
This brings seriout problems like the layer becoming distorted and take off.  - Pic att.
And the last top solid layer always is printed with zero cooling...

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 4

### Operating system

windows 10 21H2

### Printer model

ender s1

kubispe1, 2023-02-22:
Can you attach this 3mf please?

onolox, 2023-02-22:
I added the 3mf. But you can pick any shape, like a box, the last top layer will be with zero airflow.
I've used an box, the top layer uses zero fan.
I think if you use 2 or more layers not using airflow in the last "Top solid layer" generates the best finnish, but if you use only one layer it is a disaster.
I use 1 layer a lot to generate prototypes.

onolox, 2023-03-01:
> I added the 3mf. But you can pick any shape, like a box, the last top layer will be with zero airflow.
> I've used an box, the top layer uses zero fan.
> I think if you use 2 or more layers not using [...]

I added the 3mf. But you can pick any shape, like a box, the last top layer will be with zero airflow.
I've used an box, the top layer uses zero fan.
I think if you use 2 or more layers not using airflow in the last "Top solid layer" generates the best finnish, but if you use only one layer it is a disaster.
I use 1 layer a lot to generate prototypes.

onolox, 2023-03-01:
> Can you attach this 3mf please?

I added the 3mf. But you can pick any shape, like a box, the last top layer will be with zero airflow.
I've used an box, the top layer uses zero fan.
I think if you use 2 or more layers not using airflow in the last "Top solid layer" generates the best finnish, but if you use only one layer it is a disaster.
I use 1 layer a lot to generate prototypes.

closed 2026-09-29 by prusaslicer-bot[bot]