prusa3d/PrusaSlicer #9831: Randomized placement

Anothermbdusted, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
Not a problem so to speak but could be after a while to some

**Describe the solution you'd like**
I personally would like to see a thing for random placement of the model on the plate as to not wear out the center area 

**Describe how it would work**
instead of selecting auto centering in preferences there would be a choice to random placement, in doing so the model would always move around the plate different from the last one sliced.  no matter if it was small or large it would always shift a tiny bit in some way

**Describe alternatives you've considered**
manually moving a model sucks lol

**Additional context**
I just ruined a bed plate as i didnt move it around and use the outer edges enough over time.

Anothermbdusted, 2023-02-22:
I understand its not complicated at all. sometimes we get in a hurry to print something and forget to do it like i did and well it caused a issue. would just be a nice feature to have so you didnt have to worry about it. one less thing to worry about when printing and in the long run it would most likely prolong the use of the bed surface for longer use.  maybe have a no go zone where it wouldnt print in the area where the bed surface wasnt damaged and gave you a warning that you could bypass if you wanted to but in general it would avoid the areas as much as possible.

Anothermbdusted, 2023-02-22:
> It is very easy to move any model on the table. 2 clicks: 1-select the Move bodies icon on the left 2-grab one of the direction gizmos and drag that's it... Of course I'm not saying it couldn't be done, but it feels a bit overkill. It's something everyone can do themselves, and it's not complicated.

I understand its not complicated at all. sometimes we get in a hurry to print something and forget to do it like i did and well it caused a issue. would just be a nice feature to have so you didnt have to worry about it. one less thing to worry about when printing and in the long run it would most likely prolong the use of the bed surface for longer use. maybe have a no go zone where it wouldnt print in the area where the bed surface wasnt damaged and gave you a warning that you could bypass if you wanted to but in general it would avoid the areas as much as possible.

neophyl, 2023-02-22:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/320  (and several others).

kubispe1, 2023-02-22:
Closing as a duplicate. Thanks

closed 2023-02-22 by kubispe1